prusa3d/PrusaSlicer #8256: Location error

daizhuo, 2022-04-29:
### Description of the bug

I set my location in my country and can't open Prusaslicer Appimage, this is so weird. I try version below 2.1.0 made by others and they workd, but after 2.1.0 I couldn't open any version.  Version 2.2.0 told me that it could located US location, so I set my loction to US, and finally everything goes fine. I can open all version Appimage now. Could you check these for up comming reales ?

Raspberry Pi4b 4G with 64G SD Card

### Project file & How to reproduce

A

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Pi OS, 10.0 (buster)

### Printer model

i3 Mk3s+

bubnikv, 2022-04-29:
Please provide screenshots of the error messages.

You may need to install locales for the active language.

lukasmatena, 2024-02-19:
No response, closing.

closed 2024-02-19 by lukasmatena