prusa3d/PrusaSlicer #8257: Inserting pause in gcode doesn't pause the print (Pronterface-USB)

GitKap, 2022-04-29:
### Description of the bug

I inserted pause in the gcode using PrusaSlicer 2.4.2 and attempted to print via USB using Pronterface. When the printer hit given layer it displayed the message with the instruction but didn't pause the print. 

### Project file & How to reproduce

Open stl -> Create pause using PrusaSlicer right click on the Z axis -> Type message (no special characters) -> Save gcode.
Open gcode in Pronterface (v. 2014.03.10 that is delivered with Prusa installer) and start the print.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2+win64

### Operating system

Windows 10  10.0.19044 Build 19044

### Printer model

MK3s

rtyr, 2022-04-29:
It looks like firmware issue. Please provide your 3MF project file and the g-code, so I can check, whether pause is correctly added.

GitKap, 2022-04-29:
[PauseIssue.zip](https://github.com/prusa3d/PrusaSlicer/files/8591398/PauseIssue.zip)

rtyr, 2022-04-29:
M601 command (Pause) is included in the g-code, so the g-code is generated correctly by PrusaSlicer.

Please open this issue at https://github.com/prusa3d/Prusa-Firmware/issues (if similar issue is not already created there).

I am closing the issue here, since the issue is not related to PrusaSlicer.

closed 2022-04-29 by rtyr