prusa3d/PrusaSlicer #8258: MMU2 Filiment Swap without Purge

unevencube, 2022-04-29:
### Description of the bug

2 parts, set as 2 separate parts to be printed sequentially, in 2 different colors.
The first object was completed successfully in black.
The filament was unloaded, the next filament was loaded to the extruder motor, not the nozzle.
The print then resumed without completing the filament load.

### Project file & How to reproduce

[MMULoadingIssuePrintFiles.zip](https://github.com/prusa3d/PrusaSlicer/files/8592780/MMULoadingIssuePrintFiles.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2 +win-64-202201251110

### Operating system

Windows 10 Pro

### Printer model

Prusa MK3S+ and MMU2

rtyr, 2022-04-29:
Expected behavior. Sequential print with the MMU (with wipe tower) is currently not supported.

Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/1787.

Closing.

closed 2022-04-29 by rtyr

unevencube, 2022-04-29:
The wipe tower was disabled per the slicer's instructions.
The feature I was expecting was similar to what happens if you set a color change on the MK3S+ single extruder: the extruder moves off of the build plate then fully loads the next filament, then returns to the build plate to begin the next object.

rtyr, 2022-04-29:
> The wipe tower was disabled per the slicer's instructions.

Yes, so it cannot be used with MMU (which requires wipe tower for tool changes).

> The feature I was expecting was similar to what happens if you set a color change on the MK3S+ single extruder

Wipe tower is required for the tool change (ramming sequence, cleaning previous material, priming).

You can add M600 gcode to the "between object g-code" (printer settings -> custom g-code), but it will require manual filament change.

So in general, sequential print with the MMU is currently not supported.