prusa3d/PrusaSlicer #8259: Layer view disappears after slicing

RileyKnox, 2022-04-29:
### Description of the bug

When I slice a part the layer view will appear for a couple of seconds when the gcode is generating and then disappear once that's done.   The build plate goes empty and even the layer slider is missing.  If I go back to 3d viewer the part is still there though.  I have had the same issue with 2.4.1 and 2.4.2.  

This issues has happened on every file I have tried to slice with both STL and 3MF formats.  

Apparently it only happens for the Mach3 flavor, which is what my machine runs. 

I am running this on a new Dell precision laptop with 32Gb of ram and a Quatro A2000 GPU.  The computer is only a few months old.  

### Project file & How to reproduce

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2 and 2.4.1

### Operating system

Windows 11

### Printer model

Custom, not a prusa model.  Using Mach 3 gcode

rtyr, 2022-04-29:
Please provide your 3MF project file (any project where the issue occurs).

RileyKnox, 2022-05-02:
It is literally every file I have tried.

RileyKnox, 2022-05-02:
Also the system won't let me attach a 3MF file.

RileyKnox, 2022-05-02:
It's just the standard xyz cube right from thingiverse 
[xyzCalibration_cube_2x_Mag+CHT.zip](https://github.com/prusa3d/PrusaSlicer/files/8604786/xyzCalibration_cube_2x_Mag%2BCHT.zip)

RileyKnox, 2022-05-02:
Like I said.  Any file has this problem

rtyr, 2022-05-03:
Please provide your 3MF project file from PrusaSlicer. The one you provided is from Cura.

> Like I said. Any file has this problem

That is why we need your project file, so we can see what is wrong with your profiles.

lukasmatena, 2022-05-03:
[image]

You have substituted all extrusion moves from `A` (normally used by Mach3/LinuxCNC firmware) to `C`. The G-Code Viewer does not know that those are extrusions. The substitution also breaks internal tags, but since there is nothing the Viewer thinks should be previewed, it does not matter.

rtyr, 2022-10-18:
Answered. Closing.

closed 2022-10-18 by rtyr