prusa3d/PrusaSlicer #8262: Prusa Slicer 2.4.2 breaks paint on supports

CD1010, 2022-04-30:
### Description of the bug

I got 2.4.2 today and now i cannot paint on or remove support areas.  

### Project file & How to reproduce

[Nautilus PRUSA.zip](https://github.com/prusa3d/PrusaSlicer/files/8595485/Nautilus.PRUSA.zip
Try to paint supports on the tail. It will not work.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2 win 10

### Operating system

windows 10

### Printer model

Prusa i3 mks+

lukasmatena, 2022-04-30:
Quoting from [2.4.0-alpha2 change log](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha2):

> Triangle meshes imported from 3MFs, AMFs and OBJs are no longer repaired by admesh. For PrusaSlicer generated 3MFs it means that the triangle meshes will be loaded exactly the same way they were stored, while earlier PrusaSlicer may have modified a mesh with non-manifold edges slightly by trying to repair it each time a 3MF was loaded. For 3MFs not produced by PrusaSlicer and for AMFs and OBJs it may mean that model errors will not be fixed, however in contrary to the STL file format, the 3MF, AMF and OBJ file formats are indexed triangle sets with triangle neighborship clearly expressed by shared vertex indices, thus cracks in triangle meshes are less likely. Not repairing 3MFs through admesh considerably speeds up loading of PrusaSlicer project files.

I believe that this causes your issue. The model is broken and PrusaSlicer no longer repairs it during import, meaning that the painting tool does not make correct sense of it. Repairing the model by Netfabb fixed the issue for me.

CD1010, 2022-05-01:
Yes that was it.  I did netfab and was able to edit.   Thanks

closed 2022-05-01 by CD1010