prusa3d/PrusaSlicer #8264: Unwanted extrusion bug

2Kasi2, 2022-04-30:
### Description of the bug

For some reason the the printed part has two blobs of filament on 2 layers. It is visible on the slicer as thick white lines. If I rotate the part, the lines are not there:
[image]
[image]
[image]
The first picture flashes for like 1 second, thats why i missed it the first time when slicing.

I tried printing it in the first orientation twice, and it failed the first time, and the second time it finished, but it has the bulb extruded on those layers with white lines:
[image]
[image]
[image]

  

### Project file & How to reproduce

Place the file (named Reflector) in slicer and slice it, with 0.2mm layer height. I didnt adjust any settings. If i rotate the file 90 degrees, the white lines disappear. I will try printing it that way, but i dont want to loose more filament on this....
[Reflector.zip](https://github.com/prusa3d/PrusaSlicer/files/8596718/Reflector.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1 +win64

### Operating system

windows 10

### Printer model

MK3s

dartrax, 2022-05-01:
Is this a „thin wall“ where this occurs? Then probably same as 
#4495.
As a workaround, you could try to rotate the model slightly around Z.

2Kasi2, 2022-05-02:
The wall is 1.6mm thick. Rotating it works, also splitting the part, deleting top 2mm, and slicing only the bottom part works as well.

stepikovo, 2022-05-10:
I'm closing this as a duplicate of #4495

closed 2022-05-10 by stepikovo