prusa3d/PrusaSlicer #8265: Initializing StaticPrintConfigs Segmentation Fault

fraschm1998, 2022-04-30:
### Description of the bug

./PrusaSlicer-2.4.2+linux-x64-GTK3-202204251120.AppImage --loglevel 4
[2022-04-30 12:07:00.092799] [0x00007fd1960bc880] [trace]   Initializing StaticPrintConfigs
[2022-04-30 12:07:00.160514] [0x00007fd1960bc880] [debug]   full path: 10135649659260169801
[2022-04-30 12:07:00.160547] [0x00007fd1960bc880] [debug]   single instance: undefined. other params: /tmp/.mount_PrusaSBBGQtq/usr/bin/bin/prusa-slicer;--loglevel;4
[2022-04-30 12:07:00.160559] [0x00007fd1960bc880] [debug]   full lock path: /home/massimo/.config//PrusaSlicer/cache/10135649659260169801.lock
[2022-04-30 12:07:00.160597] [0x00007fd1960bc880] [debug]   Creating lockfile.
[2022-04-30 12:07:00.160606] [0x00007fd1960bc880] [info]    Instance check: Another instance not found or single-instance not set.
[1]    26737 segmentation fault  ./PrusaSlicer-2.4.2+linux-x64-GTK3-202204251120.AppImage --loglevel 4

### Project file & How to reproduce

I can't even start the application. I also tried installing from source from the gentoo repository and get the same error.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.4.2+linux-x64-GTK3-202204251120.AppImage

### Operating system

Selinux Openrc Gentoo wayland

### Printer model

Ender 3 Pro

motevets, 2022-07-24:
I am also seeing this with PrusaSlicer-2.4.2 on Arch Linux.

```
$ prusa-slicer --loglevel 4
[2022-07-24 19:15:11.564513] [0x00007f11900f0d80] [debug]   full path: 1797574042472667053
[2022-07-24 19:15:11.564527] [0x00007f11900f0d80] [debug]   single instance: undefined. other params: prusa-slicer;--loglevel;4
[2022-07-24 19:15:11.564546] [0x00007f11900f0d80] [debug]   full lock path: /home/kindrobot/.config/PrusaSlicer/cache/1797574042472667053.lock
[2022-07-24 19:15:11.564554] [0x00007f11900f0d80] [debug]   Creating lockfile.
[2022-07-24 19:15:11.564557] [0x00007f11900f0d80] [info]    Instance check: Another instance not found or single-instance not set.
zsh: segmentation fault (core dumped)  prusa-slicer --loglevel 4
```

motevets, 2022-07-24:
FWIW this seems to have been fixed in PrusaSlicer-2.5.0-alpha3.

bubnikv, 2022-07-25:
> Initializing StaticPrintConfigs Segmentation Fault

Do you have a call stack for that?

motevets, 2022-07-25:
bubnikv, should I get the call stack? I ran it with --loglevel4 and pasted the whole output.

chrishoage, 2022-08-21:
@bubnikv I am having the same issue. I posted the gdb output over here https://github.com/prusa3d/PrusaSlicer/issues/8563#issuecomment-1221604095

Please let me know if there is further information I can provide.

bubnikv, 2022-08-22:
@motevets Call stack would certainly help.

deadpickle, 2022-11-18:
Any fixes for this... I got the same for 2.50 on Ubuntu 20.04

lukasmatena, 2023-03-08:
Can you please retest in 2.6.0-alpha5? If this was the same as #8563, it should be fixed now.

lukasmatena, 2023-11-22:
No response, probably fixed. Closing.

closed 2023-11-22 by lukasmatena