prusa3d/PrusaSlicer #8266: Access Violation slicing Integrated Rewinder in Parts

OttScott, 2022-04-30:
### Description of the bug

I get an Access Violation Exception when slicing a multi-body STL saved as 3MF. I created the STL from the IntegratedRewinder_v5_6_1.scad file, imported it into PrusaSlicer 2.4.2+Win64, and Split the body into Objects. I moved half of the parts off the plate then saved this as Part2. When I attempt to slice the attached 'IntegratedRewinder_v5_6_1_Part2.3mf', I receive the Access Violation. Please see attached files.
[IntegratedRewinder_v5_6_1_Part2.zip](https://github.com/prusa3d/PrusaSlicer/files/8597669/IntegratedRewinder_v5_6_1_Part2.zip)

### Project file & How to reproduce

1. Open Attached IntegratedRewinder_v5_6_1_Part2.3mf
2. Attempt to Slice
3. Observe Exception
[IntegratedRewinder_v5_6_1_Part2.zip](https://github.com/prusa3d/PrusaSlicer/files/8597680/IntegratedRewinder_v5_6_1_Part2.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2+Win64

### Operating system

Windows 10 v21H2 (Build 19044.1645)

### Printer model

Prusa Mk3S

OttScott, 2022-04-30:
When all models are on the plate, it completes sliceing just fine.
[image]

lukasmatena, 2022-05-09:
Thanks, reproduced. It crashes when generating supports and raft. We will look into it.

bubnikv, 2022-05-13:
The support generator crashes for an object that is just very tiny. Most likely it was created by splitting a large object to multiple disconnected objects.
[nic.zip](https://github.com/prusa3d/PrusaSlicer/files/8689108/nic.zip)

lukasmatena, 2024-01-02:
This crash was already solved in 2.6.0. The project indeed contains a microscopic object.

closed 2024-01-02 by lukasmatena