prusa3d/PrusaSlicer #8267: SLA - Manual Support Placement (Pad around Object)

bwright86, 2022-04-30:
### Description of the bug

While slicing a keycap model, I am printing directly on the bed ("Pad around object" checked). When doing this, i noticed the elevated stem wasn't being supported as i would like. When i enter "Manually Editing" the SLA Support Points,

I have noticed that none of the points are attaching to the outside of my model. When I place it from a bottom-up perspective, it looks like the attach, but as I pivot I will find that the point is in mid-air between my print and some abstract point (It could be below the plate, it could be on the build plate, or 1-2 mm above the build plate).

Maybe a duplicate of #4750

### Project file & How to reproduce

[Keycap R1 BW.3mf.txt](https://github.com/prusa3d/PrusaSlicer/files/8597768/Keycap.R1.BW.3mf.txt)
1. Import the attached Keycap file
2. Check "Pad Around Object" to remove elevation (not necessary, but is the desired setting)
3. Attempt to manually add supports to underneath stem that is suspended 1.2 mm above surface.
4. Manual points will not stick to print, but will be suspended in the air.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.1+Win64

### Operating system

Windows 10

### Printer model

Photon Mono X (Resin)

stepikovo, 2022-05-09:
The issue is something different, there is too little space in between the Pad and bottom of the stem. When you reduce the Pad thickness or the Pillar diameter to 0.5 mm, it works. We will check if something can be done with it

bwright86, 2022-05-09:
Ah, I didn’t realize that, and will have to try that.

I was wondering if it had something to do with how close the stem/island was to the build plate.

Maybe if there was a detection step for supports that can’t generate because it is too close to the build plate, that could help.

closed 2026-09-29 by prusaslicer-bot[bot]