prusa3d/PrusaSlicer #8268: Slicing error?

JWartenberg, 2022-05-01:
### Description of the bug

While I'm waiting for my MK3S+ I started to work with the Prusa Slicer instead of Cura for my Ender 3 V2.

With a specific STL I have an issue with an incomplete layer. The issueappears at any printer oder slicing setting. It is visible on the print with my Ender.

The issue appears NOT with Cura Slicer and appears NOT on the print.
Due to this, I think it is an issue with the Prusa Slicer.

See attached files: STL, project generated with Prusa Slicer and screenshot.
STL source: https://www.thingiverse.com/thing:4273909

Regards
Jochen

### Project file & How to reproduce

[Slicer Issue.zip](https://github.com/prusa3d/PrusaSlicer/files/8598352/Slicer.Issue.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Prusa Slicer 2.4.2

### Operating system

Windows 10 64bit

### Printer model

Ender 3 V2

dartrax, 2022-05-01:
Can you please add/attach the screenshot? 😉

JWartenberg, 2022-05-01:
Screenshot: Tank-End_B HC.jpg in the ZIP and as attachment ;-)
[image]
[Slicer Issue.zip](https://github.com/prusa3d/PrusaSlicer/files/8598458/Slicer.Issue.zip)
Attached the ZIP again. 
.

dartrax, 2022-05-01:
Thx. I‘m on mobile phone so I could not see a pic in a zip 😉

dartrax, 2022-05-01:
I think this is the same as https://github.com/prusa3d/PrusaSlicer/issues/2434 and https://github.com/prusa3d/PrusaSlicer/issues/1448.

rtyr, 2022-05-03:
See https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ). Closing.

closed 2022-05-03 by rtyr