prusa3d/PrusaSlicer #8269: [Feature Request] Lock object position on the build plate

jakecausier, 2022-05-01:
**Is your feature request related to a problem? Please describe.**
A frustrating issue I'm having is positioning support enforcers/blockers on the build plate, but accidentally clicking onto my printable object and moving it from the position I want.

**Describe the solution you'd like**
A 'lock' option similar to the 'visible' and 'printable' options already offered. These are shown on the context menu when right-clicking the object and as a column in the objects list on the right side of the screen.

**Describe how it would work**
When clicking the lock option, the object in question will be frozen in place, where primary mouse interaction is disabled until unlocked again. Right clicking would still be permitted to allow ease-of-access.

**Describe alternatives you've considered**
Turning the item invisible or disabling printing on it does not prevent primary mouse interaction with said object.

**Additional context**
The locked object in question could turn another colour to indicate that it has been locked. Grey or red would be good options?

foreachthing, 2022-05-02:
That's why I use [Blender](https://www.blender.org/) to position modifiers and such.
One can lock an object in place (better yet: make it un-selectable (except in the outliner)).
You also get the "in front" view ("x-ray" as previously requested from PS as well) ... Blender has it all - *duckandrun* :wink:

RichardBronosky, 2022-08-26:
This would be amazing, because as it stands, you cannot rotate the view without zooming out so that you have empty space to click.

kubispe1, 2022-09-07:
Could be useful for any feature with painting tool. Nice

closed 2026-09-29 by prusaslicer-bot[bot]