prusa3d/PrusaSlicer #8271: Unable to load print preset with certain name structure

Cetarik, 2022-05-01:
### Description of the bug

I am unable to import print preset with space and 4 letters behind it: [print:0.35mm FAST]
When i import [print:0.35mm FAST], import is FAIL
When i import [print:0.05mm DETAIL], import is OK
When i import [print:0.35mm DETAIL], import is OK
When i import [print:0.35mm FASTTT], import is OK
When i import [print:0.35mm FASTT], import is OK
When i import [print:0.35mm_FAST], import is OK
Result is "0 presets successfully imported".

### Project file & How to reproduce

[print_0.35mm_FAST.ini.zip](https://github.com/prusa3d/PrusaSlicer/files/8599071/print_0.35mm_FAST.ini.zip)
[image]
File > Import > Import Config Bundle > open print_0.35mm_FAST.ini

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1

### Operating system

Windows 7

### Printer model

Prusa MK2S

Cetarik, 2022-05-01:
[print:0.20mm NORMAL] also doesnt import.

stepikovo, 2022-05-09:
The same profiles are already in PrusaSlicer, that's why when you rename them you are able to load them then.

closed 2022-05-09 by stepikovo