prusa3d/PrusaSlicer #8274: Elephant foot compensation not working properly on acute angle shape

BriLec, 2022-05-01:
### Description of the bug

I have a small part which has an acute (~45deg) corner.  Elephant foot compensation isn't applied in that corner.  For my part, I have elephant foot compensation set to 0.2mm.  I've attached two pictures showing layer 1 and layer 2, and they show that the first trace at the acute angle corner goes all the way to the edge, instead of being offset by 0.2mm.  I've tried the same operation in Cura, and I'm getting the proper offset.  Unfortunately, I cannot upload the part in question since it belongs to my customer, but the picture shows very clearly what is happening...

### Project file & How to reproduce

[EFProblemSmallAngle-Layers1-2.zip](https://github.com/prusa3d/PrusaSlicer/files/8599743/EFProblemSmallAngle-Layers1-2.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.4.2+win64

### Operating system

Windows 11

### Printer model

Prusa i3 MK3S

stepikovo, 2022-05-10:
This is intended behaviour, the Elephant foot compensation is not decreasing the size of small features to avoid eliminating them completely. A good example would be small text on the first layer.

closed 2022-05-10 by stepikovo

BriLec, 2022-05-10:
I made a sample part which I can share (attached).  Same issue.  Change the X offset of the "helper disk" from -8.18 to -8.17 to see this issue.  Same will happen if you change the Y offset from 8.18 to 8.17.   There is a scale component to this bug: This does not seem to happen on larger objects.

I had to switch away from Prusa slicer because of this issue, and I'm now using CURA, which I confirm does not suffer from this issue...  This is a real issue and not "intended behavior"...

[SamplePart-NoEFCompensation.zip](https://github.com/prusa3d/PrusaSlicer/files/8663243/SamplePart-NoEFCompensation.zip)

stepikovo, 2022-05-11:
@BriLec I'm assuming this comment supposed to be under your second issue #8275, can you put it there, please?

BriLec, 2022-05-11:
Hi stepikovo,
This new file applies to both issues.  When changing the X/Y position of the "helper part" as indicated in the post, the elephant foot compensation is not applied to the main object in some cases (main object is the triangle part in this case).  Note that even if the triangle part is 10x bigger, the same issue is still present (I can upload another project file if you want).
Because of the 45deg angles, it is also possible to see the 2nd issue.
RegardsBrian