prusa3d/PrusaSlicer #8276: libslic3r "Mutable priority queue - first pop" test failure

craftyguy, 2022-05-02:
### Description of the bug

I'm working to package this application on Alpine Linux and this test fails:

```
❯ /home/clayton/src/aports/testing/prusa-slicer/src/PrusaSlicer-version_2.4.2/build/tests/libslic3r/libslic3r_tests "Mutable priority queue - first pop"
[2022-05-02 11:39:46.025235] [0x00007f9b4015fc20] [trace]   Initializing StaticPrintConfigs
Filters: Mutable priority queue - first pop
Testing Mutable priority queue - first pop

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libslic3r_tests is a Catch v2.13.6 host application.
Run with -? for options

-------------------------------------------------------------------------------
Mutable priority queue - first pop
-------------------------------------------------------------------------------
/home/clayton/src/aports/testing/prusa-slicer/src/PrusaSlicer-version_2.4.2/tests/libslic3r/test_mutable_priority_queue.cpp:343
...............................................................................

/home/clayton/src/aports/testing/prusa-slicer/src/PrusaSlicer-version_2.4.2/tests/libslic3r/test_mutable_priority_queue.cpp:367: FAILED:
  CHECK( valid )
with expansion:
  false

===============================================================================
test cases: 1 | 1 failed
assertions: 1 | 1 failed
```

### Project file & How to reproduce

Reproducing this requires a system running Alpine Linux (or a container). A preliminary APKBUILD + patches required to build it are located here: https://gitlab.alpinelinux.org/alpine/aports/-/merge_requests/33890

It can be used to build/run tests with `abuild -r` from the directory where the APKBUILD + patches are located.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Alpine Linux (edge)

### Printer model

n/a

Jony01, 2022-05-06:
Hi @craftyguy,
could you please confirm, that this fix helps solve your issue with test:
https://github.com/prusa3d/PrusaSlicer/commit/6ab517187fa5f3a8e2b87a59b6a50f26b3a49da6

kk6mrp, 2022-05-12:
It still fails for me:

<details>
<summary>Details</summary>

```
2/5 Test #2: libslic3r_tests ..................***Failed   19.11 sec
[2022-05-12 12:44:37.062442] [0x00007fda98ef5760] [trace]   Initializing StaticPrintConfigs
Testing sort_remove_duplicates
Passed in 1.3e-05 [seconds]

Testing string_printf
Empty format with empty data should return empty string
String output length should be the same as input
String format should be interpreted as with sprintf
String format should survive large input data
Passed in 7.4e-05 [seconds]

Testing Scenario: Reading 3mf file
    Given: umlauts in the path of the file
     When: 3mf model is read
     Then: load should succeed
Passed in 0.001067 [seconds]

Testing Scenario: Export+Import geometry to/from 3mf file cycle
    Given: world vertices coordinates before save
     When: model is saved+loaded to/from 3mf file
     Then: world vertices coordinates after load match
Passed in 0.021493 [seconds]

Testing Scenario: 2D convex hull of sinking object
    Given: model
     When: model is rotated, scaled and set as sinking
     Then: 2D convex hull should match with reference
Passed in 0.002775 [seconds]

Testing Building a tree over a box, ray caster and closest query
Passed in 3.7e-05 [seconds]

Testing Scenario: Constant offset
    Given: 20mm box
     When: Slic3r::offset()
plus 1mm, miter 2x
     Then: Area is 22^2mm2
    Given: 20mm box
     When: Slic3r::offset()
minus 1mm, miter 2x
     Then: Area is 18^2mm2
    Given: 20mm box
     When: Slic3r::offset()
plus 1mm, miter 1.5x
     Then: Area is 22^2mm2
    Given: 20mm box
     When: Slic3r::offset()
minus 1mm, miter 1.5x
     Then: Area is 18^2mm2
    Given: 20mm box
     When: Slic3r::offset()
plus 1mm, miter 1.2x
     Then: Area is 22^2mm2
    Given: 20mm box
     When: Slic3r::offset()
minus 1mm, miter 1.2x
     Then: Area is 18^2mm2
    Given: 20mm box
     When: Slic3r::variable_offset_outer/inner
plus 1mm, miter 2x
     Then: Area is 22^2mm2
    Given: 20mm box
     When: Slic3r::variable_offset_outer/inner
[... 914 lines cut ...]

Testing Hollow two overlapping spheres
Passed in 3.55225 [seconds]

===============================================================================
test cases:    88 |    87 passed | 1 failed
assertions: 48233 | 48231 passed | 2 failed

    Start 3: slic3rutils_tests
3/5 Test #3: slic3rutils_tests ................   Passed    0.02 sec
    Start 4: fff_print_tests
4/5 Test #4: fff_print_tests ..................   Passed    6.93 sec
    Start 5: sla_print_tests
5/5 Test #5: sla_print_tests ..................   Passed   63.90 sec

80% tests passed, 1 tests failed out of 5

Total Test time (real) =  90.43 sec

The following tests FAILED:
	 2 - libslic3r_tests (Failed)
Errors while running CTest
```

</details>

craftyguy, 2022-05-12:
and just in case there's some more context that might be useful, here's the test failure in the Alpine Linux CI:

https://gitlab.alpinelinux.org/craftyguy/aports/-/jobs/719734#L4981

kk6mrp, 2022-05-13:
There are no test failures with patches to boost. Refer to https://github.com/prusa3d/PrusaSlicer/pull/6849#issuecomment-1125598262

Boost 1.77 needs the following patches for Prusa Slicer to compile correctly:
https://github.com/boostorg/math/commit/3d3d2d7467cf9e0e8dc3220cf1a2badb321df099.patch
https://github.com/boostorg/geometry/commit/5110ec7daea0fef72ebdbd41c7f3392502c55945.patch
https://github.com/boostorg/geometry/commit/3dbd2247b63613f68ed4feebb5f5ba390d021b39.patch

craftyguy, 2022-05-13:
@kk6mrp hmm, the test is still failing even with the patches to boost: https://gitlab.alpinelinux.org/craftyguy/aports/-/jobs/720012#L1911

so I'm curious how/why it passed for you 😄

kk6mrp, 2022-05-13:
@craftyguy I have no idea, I built it successfully three or four times and then I've got the same issue as you now.

craftyguy, 2022-05-13:
@Jony01 so I think it's fairly safe to say at this point that your patch doesn't resolve the failure for us. Is there anything I can collect to help?

bubnikv, 2022-05-17:
@craftyguy would you please test the current master?

I hope I have fixed the unit tests with
15a082b80b1d9ed3cfa70d3d8133c4cd1e5d699d
f5ec76c2300095f23af6be9d37488b7f434df25d

craftyguy, 2022-05-28:
I'm having some unrelated issues trying to build master.. and it looks like those patches are not trivial (to someone unfamiliar with the codebase) to backport to the 2.4.x stable branch, which is what I've been trying to package all along. Any chance these patches could be backported to the latest stable release?

kk6mrp, 2022-07-28:
@bubnikv I can build on the latest master but some of the tests are still failing:
```
Test project /home/weston/dev/aports/testing/prusa-slicer/src/PrusaSlicer-master/build
    Start 1: libnest2d_tests
1/5 Test #1: libnest2d_tests ..................   Passed    0.54 sec
    Start 2: libslic3r_tests
2/5 Test #2: libslic3r_tests ..................Subprocess aborted***Exception:   0.07 sec
[2022-07-27 23:52:18.468394] [0x00007fa796a75be0] [trace]   Initializing StaticPrintConfigs
Assertion failed: scaling_factor.x() > 0.0 && scaling_factor.y() > 0.0 && scaling_factor.z() > 0.0 (/home/weston/dev/aports/testing/prusa-slicer/src/PrusaSlicer-master/src/libslic3r/Geometry.cpp: set_scaling_factor: 555)

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libslic3r_tests is a Catch v2.13.6 host application.
Run with -? for options

-------------------------------------------------------------------------------
Scenario: Export+Import geometry to/from 3mf file cycle
      Given: world vertices coordinates before save
-------------------------------------------------------------------------------
/home/weston/dev/aports/testing/prusa-slicer/src/PrusaSlicer-master/tests/libslic3r/test_3mf.cpp:27
...............................................................................

/home/weston/dev/aports/testing/prusa-slicer/src/PrusaSlicer-master/tests/libslic3r/test_3mf.cpp:27: FAILED:
due to a fatal error condition:
  SIGABRT - Abort (abnormal termination) signal

===============================================================================
test cases: 4 | 3 passed | 1 failed
assertions: 7 | 6 passed | 1 failed

    Start 3: slic3rutils_tests
3/5 Test #3: slic3rutils_tests ................   Passed    2.55 sec
    Start 4: fff_print_tests
4/5 Test #4: fff_print_tests ..................   Passed   40.02 sec
    Start 5: sla_print_tests
5/5 Test #5: sla_print_tests ..................   Passed   63.78 sec

80% tests passed, 1 tests failed out of 5

Total Test time (real) = 106.96 sec
```

kk6mrp, 2022-08-07:
Any ideas?

closed 2026-09-29 by prusaslicer-bot[bot]