prusa3d/PrusaSlicer #8277: Wrong temperature selection on layer(1)

tommylux, 2022-05-02:
### Description of the bug

With 3 extruders enabled;
The lowest numbered extruder printers first (1) with a temperature selection for other layers set to 250;
No tool changes between layer 0 and layer 1
The GCode ouput seems to select another filaments value other than for the first layer.
[image]

### Project file & How to reproduce

[Flurry and Evie Rabbits.zip](https://github.com/prusa3d/PrusaSlicer/files/8605608/Flurry.and.Evie.Rabbits.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1

### Operating system

Win 10

### Printer model

A20T

lukasmatena, 2022-05-03:
Tools indices are zero/based. Lowest number extruder is 0, and its temperature does not change, it is set to 250/250. On the other hand, the second extruder (1) is set to have 220 on second and higher layers, which is what this command does. The first tool (numbered zero) will stay at 250.

Handling of temperature changes for multi-extruder printers can indeed be improved, for example by only heating extruders before they are actually used. We are already discussing that. But the behavior you described is not wrong. Does it answer the question?

tommylux, 2022-05-03:
The printer I have is a shared hot end. It was heating T1 whilst printing on T0, but both commands change the temperature on the same hotend.

tommylux, 2022-05-03:
It was also heating T1 several layers before it finished printing on T0 meaning that T0 printing PETG was printing too cold.

lukasmatena, 2022-05-03:
If by "shared hotend" you mean that there is just one nozzle and some mechanism that feeds different filaments into it (much like Prusa MMU unit does), it would be polite to also tell it to PrusaSlicer by checking "Single Extruder Multi Material" in `Printer Settings->General`.

tommylux, 2022-05-03:
I understand that I need to have that setting unselected, because it introduces a retract as if it was a switching hotend, rather than a mixing one?

I would be unsure what to set in the MM setup.

I have a mixing hot end, please advise.

Thanks

Tom

closed 2026-09-29 by prusaslicer-bot[bot]