prusa3d/PrusaSlicer #8278: GUI Text size

mullcom, 2022-05-02:
**Is your feature request related to a problem? Please describe.**
Yes and now. I am sending in this request by i have some problem with the Text in slicer application. Its to small and want to make it bigger. its hard to see the text on the screen sometimes.  I also notes when i change mouse icon to bigger in windows i get text behind the icon and i cant read the text Property 

**Describe the solution you'd like**
 Have ability to change text layout so it fits resolution on the monitor or bad seeing. yast have ability to change size.

**Describe how it would work**
in preference  have option to change text size of Tooltips or menu  or settings on the side panel  and so on.

**Additional context**
Its hard for me to read the text its to small mostly everywhen.
[image]
[image]
[image]
[image]
[image]

lukasmatena, 2022-05-03:
PrusaSlicer should use your system font size, changing font scaling in system settings should help. Or do you say that all applications use big fonts and only PrusaSlicer small ones? Similar to #6317.

>  I also notes when i change mouse icon to bigger in windows i get text behind the icon and i cant read the text Property

I'm not sure I understand, but it may be a duplicate of #7785, which was already fixed for the next big release.

mullcom, 2022-05-03:
Hi. The font is okey in windows. As you can see it's a lot of free space that can be used.  Sometimes i use bigger text in application then the main windows. 

And the tooltips is neatly impossible yo read. I can send a screenshot from whare i am sitting so you see my point why there are intrest to have this settings.

mullcom, 2022-05-03:
Here comes a view from end user that using the application
[image]
.

PilotiE36, 2022-07-01:
I concur with the OP, as well as the comments in #6317 .

This isn't merely an issue of text size, but an ill-proportioned overall UI and poor utilization of screen space, and can't be solved merely by setting a larger system font size, which isn't a solution in itself.

The screenshots of the settings windows illustrate the issues well.  The UI elements are scrunched into the top 30% and left 50% of the screen area, with tiny, hard to read fonts, with vast expanses of barren, underutilized space.

PrusaSlicer is my slicer of choice, and I'm grateful for the app, and the developers for giving us the option, but these issues present a hindrance to usage, and does not set a good example of UX.

Vision loss in the normal course of aging will affect every person, and a lot of the UX developers who, in their own form of myopia, design interfaces only for younger eyes will come to regret their choices when they, too, age and eventually experience it.

mullcom, 2022-07-01:
> I concur with the OP, as well as the comments in #6317 .
>
> This isn't merely an issue of text size, but an ill-proportioned overall UI and poor utilization of screen space, and can't be solved merely [...]

Thanks for your comment. You really explain it well. especially age problems. I am not Ung anymore and need to convince my self it's starting be a problem. Even i have always have good sight. 

I really like prusa slicer and what you do for the community. What dose not are in the screenshot is the new tooltips in bottom right. This is also hard to read. For me i skipping to read it because of that.

Thanks for everything.

jel111, 2022-09-12:
I'm getting older, and as PilotiE36 stated, I wish we could at least use command plus a shortcut to increase size. It's so damn small I have to use Ideamaker even though I would rather use PrusaSlicer..

benbelly, 2023-01-17:
Due to severe myopia, I have some early, mild macular degeneration. I deal with this by scaling font sizes in most applications a bit. I prefer this to using the system font size, because in Windows 11 the applications respond inconsistently, but consistently poorly. Fonts I don't care so much about are scaled (e.g., my tab title fonts), but the fonts I want scaled (e.g., PrusaSlicer everywhere) seem to ignore the Accessibility Text Size scaling. Those things that do scale ARE ok but look bad and tend to be less usable than those things that make a point of implementing and testing their own app to maintain usability with scaling.

ronguest, 2023-01-29:
I recently switched from Cura to PrusaSlicer on macOS. I use a 27” iMac (5K display). I immediately noticed the text in PrusaSlicer settings was harder to read. I changed the default for text to white which, at least for me, helped since PrusaSlicer uses dark mode. That’s logical since my system is set for Dark Mode but I find the text in some apps, like Notes, hard to read due to the lack of contrast. Fortunately some apps allow the user to set whether to use the system default or to force light or dark. I think this would help PrusaSlicer readability. Secondly I compared it with Cura and I  find Cura is much easier to read. See the attached comparison. Headings in Cura are much larger. Item text in Cura certainly appears larger as well. Cura uses light mode only so I couldn’t compare what it looks like in dark mode.

I would like to see this improved as well. I also compared PrusaSlicer to apps like Messages and again the slicer’s font is definitely smaller.

[image]

lukasmatena, 2023-10-23:
Variable UI font size is implemented in [PrusaSlicer 2.7.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-alpha1). Closing.