prusa3d/PrusaSlicer #8279: Modifying a filament preset thinks your switching presets

tommylux, 2022-05-03:
When you have modified a preset, in this example Green, but you do not wish to save the settings as it is temporary for the project, when you try and modify another filament, in this case Silver, Prusaslicer forces you to either Save or discard the green preset.
[image]

**Describe the solution you'd like**
To be able to modify all filaments without saving them for the purpose of the project.

**Describe how it would work**
When you close the project, it could ask you if you want to save presets then, or when you switch a filament which is not selected as part of the extruder.

**Describe alternatives you've considered**
When turning off the GUI option to disable this notification, the default behavior becomes 'discard' and the settings are reverted, which doesn't allow you to keep modifications during the project.

**Additional context**
[image]
[image]

neophyl, 2022-05-03:
There is only one filament editing window.  So at the moment you have to save your changes to the profile before you can edit a second filament or the changes are lost.  To do what you want I think this would mean there would have to be multiple instances of the filament editing tab/window opened.  One for each filament dropdown configured.  Not sure on if that's a easy or difficult change for the dev's though.

tommylux, 2022-05-03:
Yes, you understand my point.

Either a new window for x number of extruders, or, if you could save the modifications into the project file, this would achieve the same thing. (Given that if you do not save presents current, they are saved in the project file). The difficulty is, that Prusa wants to either hard save the presets or revert them when switching.

Tom

lukasmatena, 2026-09-01:
Implemented in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena