prusa3d/PrusaSlicer #8280: PrusaSlicer: Slicing Orientation

seanfrisbey, 2022-05-03:
**Is your feature request related to a problem? Please describe.**
Please see the attached GCODE for reference. When slicing this project (and others in the set), the corners are printed bit by bit, back to front. As a result, if you skim through the GCODE in the viewer, you'll notice that it hops around a LOT from corner to corner and it works it's way "down" the model. This causes a bunch of travel, and results in many of the bits not sticking.

**Describe the solution you'd like**
I would really like a way to "rotate" the slicing direction. I don't really know what else to call it, but ideally it would results in a single corner printing before moving to the next.

**Describe how it would work**
Very similar to setting infill rotation, you could set slicing rotation: by degrees. This would effectively allow you to set which side or corner it starts on.

**Describe alternatives you've considered**
The alternative I have settled on is to rotate the piece. This is very easy for small models, but this particular model is 238.6 mm wide, which doesn't give me a lot of room to rotate.

**Additional context**
[GCODE Files (ZIP)](https://github.com/prusa3d/PrusaSlicer/files/8609170/GCODE.files.zip) This is a print with the MMU2s. I have sliced it so that it prints the gold inlay first, necessary to get a clear extrusion, otherwise the primary color hogs up the space and ruins the effect. To see this issue, look at each of the GCODE files, scroll down to the first layer, and start scrolling form the beginning of the layer. When printing the inlay, the "standard" slice hops back and forth a bunch. This causes dragging and jerking, and never results in bits that stay on the bed. The "rotated" slice, since it prints objects from back to front, manages to print the entire corner's contents before moving to the next one. This reliably produces perfect results every time.

lukasmatena, 2022-05-03:
Thanks for the report. Could you please also send a 3MF of the same project?

seanfrisbey, 2022-05-03:
Sure! Of both versions?

lukasmatena, 2022-05-03:
Yes, please.

seanfrisbey, 2022-05-04:
Here ya go

[3MFs.zip](https://github.com/prusa3d/PrusaSlicer/files/8617370/3MFs.zip)
!

closed 2026-09-29 by prusaslicer-bot[bot]