prusa3d/PrusaSlicer #8281: Rotate Issue after Arrange

aclucca, 2022-05-03:
### Description of the bug

If you use the arrange function and then rotate the Y-Axis 180 degrees, there is an offset.
See attached Pictures - Picture1 is a full filled bed, and if you rotate the item it is outside of the bed (picture2)

[image]
[image]

### Project file & How to reproduce

See above

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.4.2+win64

### Operating system

Win 10 64 bit

### Printer model

Prusa MK3

neophyl, 2022-05-03:
And why is this a bug ?  Click arrange again.  By rotating it you have manually changed its position/orientation.  What's so surprising about it ? The centre of rotation for the model doesn't change just because its been arranged or moved.   The function is  not called Auto Arrange.

aclucca, 2022-05-03:
because the centre of the model should remain the centre of the model, whether it was arranged before or not . . .

neophyl, 2022-05-04:
Erm it has remained the same.  The centre of rotation for the object is NOT the centre of the bed, its the centre of the OBJECT.  It has always been such.  Defining the rotation axis as the middle of the bed would be a nightmare for choice for transforms.  
As such when you manually rotated it you changed its position and moved the bounds outside the bed.  

Would you consider it a bug if after arranging some object you manually dragged it off the plate and it gave the same error ?  
Its not an issue as you manually changed the position by rotating it.  If you want it to arrange again then just click the button again.

aclucca, 2022-05-04:
Ok, thanks for the explanation.

I still have a comprehension problem. If I arrange the object on the bed
with Arrange, the center of the object should also be in the center of the
bed. That means if I rotate it 180 degrees, the center should not change.
How do you explain this?

Übersetzt mit DeepL (https://www.deepl.com/app/?utm_medium=ios-share)

neophyl, 2022-05-04:
The arrange function uses the bounding box area of an object it seems.  However an object centre is not always in the centre of the bounding box.  It depends on the object and the software you used to create it I think.  I can use something like Blender to change the centre of the object and it will use that to rotate around. 

I think I have that correct.

aclucca, 2022-05-04:
ok, but your statement confirms the bug

look at picture 1, there the object was arranged with arrange - the
bounding box is arranged centrally at the bed.

if you rotate the object by 180 degrees, the bounding box should also be
centrally located on the bed again, but it is not. see picture 2

neophyl, 2022-05-04:
No you are missing my point.  The arrange uses the bounding box.  Rotation uses the Object centre which are necessarily the same thing.   
Looking at your object from the picture you posted then I would estimate its natural centre to be between the words/lines, roughly around the bottom of the l in Familie and the top of the b in Dob.  If that is the centre of the object then just look where its is on the bed when you have arranged it.  Its not in the centre of the plater its off to the right.  So when you rotate the object its moving around that point.  rotating 180 degrees means more of the letters are now on the right than the left.  Which is exactly what you have pictured in your second picture.  Basically its doing exactly what I would expect the software to do.  That's with the centre of rotation in the centre of the actual object.  Without any modelling offset as sometimes happens,

Zip up and attach your object or better yet your saved project file and someone can examine it, but if you select it with the move or rotation tool you will be able to see where the centre of the object is.

aclucca, 2022-05-05:
hi!

i´ve selected it with the rotation tool and the middle of the object is the middle of the bed . . . 
after rotation the middle of the object isn´t anymore in the middle of the bed . . . 

see pictures
also projectfile attached

[pf.zip](https://github.com/prusa3d/PrusaSlicer/files/8632617/pf.zip)
[image]
[image]

neophyl, 2022-05-06:
When trying to extract your pf.zip it asks for a password.

aclucca, 2022-05-06:
sorry, attached new file

aclucca, 2022-05-06:
[ds2.zip](https://github.com/prusa3d/PrusaSlicer/files/8638089/ds2.zip)

kubispe1, 2023-02-12:
Hi, I think that this happened in 2.5.0. In 2.6.0 alpha3 it works as you expect. I attach gif and I am closing issue. Thanks
[image]

closed 2023-02-12 by kubispe1