prusa3d/PrusaSlicer #8282: Associate .3mf files with PrusaSlicer in Linux

yossarianmtu, 2022-05-03:
Ubuntu 20.04.4
DE: MATE (if that matters)
PrusaSlicer 2.4.2 from the repos.

**Problem**
Double clicking a .3mf file in linux results in an archive manager opening. in my case Archive Manager/ File Roller. I would like it to open with PrusaSlicer.
If I right click on the file and use the Properties -> Open With tab to change the default program to PrusaSlicer ALL compressed file types (eg. .zip or .rar) try to open with PrusaSlicer which obviously is not wanted. 

**Describe the solution you'd like**
I would like to see an option for PrusaSlicer to be the default program for .3mf files in Ubuntu/linux without assocaiting all acrchive type files with PrusaSlicer.

**Describe how it would work**
I am not familiar enough with linux to know if this is an issue on Ubuntu's side or PrusaSlicer's. I think it has something to do with [MIME Types.](https://help.ubuntu.com/community/AddingMimeTypes)

**Describe alternatives you've considered**
Obviously I could right click on the file and Open With PrusaSLicer every time, but its annoying. 

If this needs to be reported to Ubuntu please let me know. If there is another relatively easy way to make .3mf files open with PrusaSlicer please let me know as well.

yossarianmtu, 2022-05-04:
I found a user with a similar issue and fix. Perhaps it can be modified to work with PrusaSlicer? I dont understand the block of code enough to modify it for PrusaSlicer.
[https://askubuntu.com/questions/1286360/set-default-application-by-extension-not-mimetype](https://askubuntu.com/questions/1286360/set-default-application-by-extension-not-mimetype)

yossarianmtu, 2022-05-05:
The ubuntu guys helped me out. You can find the solution that worked for me here:
[https://answers.launchpad.net/ubuntu/+source/xdg-utils/+question/701652](https://answers.launchpad.net/ubuntu/+source/xdg-utils/+question/701652)
It would be great if this modification was added.

bubnikv, 2022-05-05:
```
Okay, this might work. It seemed to work in my VM:
Make a file named ~/.local/share/mime/packages/application-x-3mf.xml with vim or whatever text editor you like:
Copy and paste:
<?xml version="1.0" encoding="UTF-8"?>
<mime-info xmlns="http://www.freedesktop.org/standards/shared-mime-info">
  <mime-type type="application/x-3mf">
     <comment>3D Manufacturing Format</comment>
     <icon name="application-x-3mf"/>
     <glob-deleteall/>
     <glob pattern="*.3mf"/>
  </mime-type>
</mime-info>
Then run:
update-mime-database ~/.local/share/mime
And finally, edit ~/.config/mimeapps.list and add the following line to the [Added Associations] section:
application/x-3mf=prusa-slicer.desktop
```

Thanks for the hints. Even our in-house Linux experts did not know.

In case of distro package, snap or flatpack, it is the package responsibility to do the registration.

In case of AppImage, we have already a manual desktop integration implemented into PrusaSlicer done by @kocikdav. I asked David to implement the MIME registration in addition.

closed 2026-09-29 by prusaslicer-bot[bot]