prusa3d/PrusaSlicer #8283: In addition to custom/paint-on supports, PrusaSlicer needs custom/paint-on brim.

spasecookee, 2022-05-03:
**Is your feature request related to a problem? Please describe.**
There are occasions where only part of a model really needs a brim, and since brim removal can introduce problems and leave unsightly marks on a model, it would make sense to only have a brim where absolutely needed. I am not talking about "mouse ears"  I am talking about situations where some parts of a model have a small contact patch.

**Describe the solution you'd like**
I would like an two options, very similar to supports. One option would be a custom brim using a shape the same way the "add support enforcer" works now. Except in this case, instead of enforcing, a brim is added that uses the pre-configured brim width in print settings. The other option would be a paint-on brim that works similarly to paint-on supports.

**Describe how it would work**
The options could appear in the context menu when right-clicking a model and also as an icon on the left of the Plater window.

**Describe alternatives you've considered**
There are no good alternatives aside from using brim on every part of the model that contacts the print bed. An alternative would be a "brim-blocker" to block creation of brim, similar to how "support blocker" works. The only final option is to somehow integrate the brim into the STL file before it is introduced to prusa slicer, but this would be an advanced activity compared to it built into Prusa Slicer. Adding a cylinder to the platter is not a good solution for irregularly shaped models. None of the solutions [here](https://forum.prusa3d.com/forum/original-prusa-i3-mk3s-mk3-how-do-i-print-this-printing-help/how-to-print-brim-ears/) are as convenient or user-friendly as the existing options that have been added for supports.

lunatic1972, 2022-05-05:
This does seem like a really good one. I've printed out some articulated models before and this would have come in handy on certain parts of the model rather than doing the whole or nothing. Cleanup would have required little effort but instead it took about 45 minutes to get it cleaned up.

spasecookee, 2022-05-05:
> . I've printed out some articulated models before and this would have come in handy on certain parts of the model rather than doing the whole or nothing

This was my exact same reason for wanting this.

altuor, 2022-07-08:
strong upvote on this one, submitted similar request as I didn't see it ;-)

Exerqtor, 2022-08-06:
Upvote for this one!

iBral, 2022-08-28:
Good, I need this feature, SS have this feature.

gudvinr, 2022-09-30:
@thomas-christiansen next time you want to support some feature, just react to top message. You just notified all subscribers of this issue for no reason (as I did right now too, sorry).

Titlest, 2022-11-06:
Is your feature request related to a problem? Please describe.

FEATURE REQUEST
Sometimes when I do large prints that are filling up the print bed and also have some complex shapes and together with the usage of "brim" it is a big problem when cleaning parts after printing and it also is time-consuming. I need the brim on the parts so that they don't warp and detach from the build plate. sometimes even a 10 mm brim is needed.
Imagine a print speed of 15mm/s on the first layer and you have a complex shape that has a lot of corners but only a few of them need brims to be able to stick to the print bed. One round trip for the brim can be as an example, 500mm which will have a print time of 33 seconds (500mm/15mm's = 33) and that will be 0.5mm wide per revolution. 10mm brim will then have 20 revolutions which will be a total of 11 minutes ( 33sec'rev * 20 rev = 660sec / 60 sec=11min ) just for printing the brim. It also uses a lot more plastic than needed.

Describe the solution you'd like
For me, the solution is to just use a brim on the corners of a part where I see the problems for warping may happen.
To save material and print time on large prints I am today modeling my own brim where I see that it is needed. "see picture"

Describe how it would work
Where we have the "Paint on support" and "Seem painting" I would like to see "place brim"
When active you will be able to set the size of your brim to a radius and then click on corners from the model that is on the build plate. This will add a brim to the corner with the specified radius.

Describe alternatives you've considered
Keep on modeling the brims manually in CAD and do not use the brim function in the Slicer

[image]

sschwartzman, 2023-03-03:
Strong upvote on this one. There have been many times when I just want to print little ears to keep the corners down of a large thing, without a full brim since most of the print doesn't need them and then its more to have to remove/clean later.

Mirarkitty, 2023-08-04:
I so need this. Upvote

Abrynos, 2023-08-06:
I've recently printed a few parts that have really good bed adhesion for most of it, but extremely bad adhesion in two corners as there is very thin "arms" extending out. However, enabling brims for the whole part was very excessive so I have to strongly express my support for this feature request.

samuelT2, 2023-08-20:
Upvote!

Peacem4kr, 2023-10-05:
need :D

BernhardFe, 2023-10-13:
I strongly request the same and closed #3358. I'm  printing tools with Prusa PC and there is no chance to print them without BRIM. But it cost me a lot of time and work **(3 minutes per part and printing 200/month = 10h)** to deburr the brim away even with optimised gap between part and brim parameter. 
Doing it manually with the CAD system for a lot of parts on the print bed would be very inconvenient and time consuming.
This is waisted time. That's why I support the brim painting and as well the corner brims very strongly. I'm wondering why this is not implemented. Please, Prusa Slicer developers put it on a high priority. I would appreciate it very much... :-)

jon-whit, 2023-12-14:
I've started a pull-request to introduce this feature. It will probably take me longer than the Prusa team could do it if they focused on this effort, but I do plan on completing the feature and hopefully working with the PrusaLabs team to get it merged into PrusaSlicer. I need this for some hobby work that I do and so I figured I'd just contribute it.

Snuff1eupagus, 2024-04-28:
Not paint on but you should at least be aware that you can do this.

https://www.youtube.com/watch?v=dhsbMmosbLU&t=1s

Why this guy waited almost 3 years to tell us about this I don't know. But why didn't anyone else figure this out?

rcmaniac25, 2024-04-28:
That's quite nifty

hapiel, 2025-01-19:
> Not paint on but you should at least be aware that you can do this.
> 
> https://www.youtube.com/watch?v=dhsbMmosbLU&t=1s
> 
> Why this guy waited almost 3 years to tell us about this I don't know. But why didn't anyone else figure this out? thesse are even more capable then paint-ons could be.

Link no longer works, but here is a new one: https://www.youtube.com/watch?v=IEKUpZ-gjMs

Thynix, 2025-04-01:
> Link no longer works, but here is a new one: https://www.youtube.com/watch?v=IEKUpZ-gjMs

That video is now private. Do you have another link?

choffee, 2026-01-09:
Related to #2954

Tupson444, 2026-09-18:
still relevant

Ioann44, 2026-09-18:
Orca has it (called "mouse ears brim").  I see no reason for anyone to use prusa slicer still

Jan-Soustruznik, 2026-10-19:
Hello, this feature request seems very similar to #2680. I am closing this issue as a duplicate so we can keep the conversation and tracking in one place.

closed 2026-10-19 by Jan-Soustruznik