prusa3d/PrusaSlicer #8284: PS uses XWayland and ignores GDK_BACKEND on Wayland

gudvinr, 2022-05-04:
### Description of the bug

When used in Wayland session, PS uses XWayland.
It does so by explicitly overriding `GDK_BACKEND` environment variable.  
It is stated [here](https://github.com/prusa3d/PrusaSlicer/blame/b88fc2896e706136be12a9f765004d64d358c79e/src/PrusaSlicer.cpp#L77) that wxGTK doesn't support Wayland, which it does [since 2013](https://github.com/wxWidgets/wxWidgets/commit/659991ac34ee060e77dfd1ed503da54f8a773194).

There _were_ some issues with wxWidgets not working properly under Wayland over the years so this stance should be revisited at least.

In any case, "fixing" this by overriding user-defined environment variables can be considered a bad programming practice.

Here's proof of it being run under XWayland.
[image]

From practical standpoint, that leads to blurry fonts when fractional scaling is used:  
[image]

### Project file & How to reproduce

Issue isn't related to any particular project.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Linux (Arch Linux, KDE Plasma 5.24, GTK3 3.24.33)

### Printer model

Not applicable

lukasmatena, 2022-05-04:
The override was added as a response to https://github.com/prusa3d/PrusaSlicer/issues/4691, https://github.com/prusa3d/PrusaSlicer/issues/2515. The combination of wxWidgets+GTK3+Wayland crashed, which also some consider to be a bad practice. And those issues are not that old.

Did you compile PrusaSlicer yourself? With what wxWidgets version? It is possible that just a certain combination is affected. It is also possible that the situation has a better fix.

gudvinr, 2022-05-04:
I just tried to build PS without `GDK_BACKEND` override and wxgtk3 3.1.6 (which is kinda sorta stable-ish but not included in any distributive).

It doesn't break while calling X11 methods anymore as in #4691 but now I see segfault that comes somewhere from PS when it calls `SetLanguage`:
```
                #0  0x00005632572770d5 _ZNSt7__cxx1112basic_stringIwSt11char_traitsIwESaIwEE9_M_assignERKS4_ (prusa-slicer + 0xa540d5)
                #1  0x00007f1e53f0f2eb _ZN14wxTranslations11SetLanguageE10wxLanguage (libwx_baseu-3.1.so.6 + 0x18b2eb)
                #2  0x0000563257267856 n/a (prusa-slicer + 0xa44856)
                #3  0x00005632572696ac n/a (prusa-slicer + 0xa466ac)
                #4  0x000056325726c27b n/a (prusa-slicer + 0xa4927b)
                #5  0x00007f1e53eb3412 _Z7wxEntryRiPPw (libwx_baseu-3.1.so.6 + 0x12f412)
                #6  0x00005632572506be n/a (prusa-slicer + 0xa2d6be)
                #7  0x0000563256b2b812 n/a (prusa-slicer + 0x308812)
                #8  0x0000563256b0fb0a n/a (prusa-slicer + 0x2ecb0a)
                #9  0x00007f1e526f0310 __libc_start_call_main (libc.so.6 + 0x2d310)
                #10 0x00007f1e526f03c1 __libc_start_main@@GLIBC_2.34 (libc.so.6 + 0x2d3c1)
                #11 0x0000563256b23e55 n/a (prusa-slicer + 0x300e55)
```

lukasmatena, 2022-05-04:
@gudvinr I believe that @YuSanka has also seen the `SetLanguage` segfault when she attempted to build against wxWidgets 3.1.6, it is related to something that has changed in wxWidgets between 3.1.5 and 3.1.6. Not related.

If we knew when the Wayland issue disappeared (with which wxWidgets version), maybe we could do the env variable override conditionally. The two issues were reported by package maintainers who link against 3.0.x.

lukasmatena, 2022-05-04:
@gudvinr We are currently during the process of updating from 3.1.4 to 3.1.6. We will test whether the override is still necessary. There may be other issues, even if the crash is gone. If all is fine, we would do something like this: https://github.com/prusa3d/PrusaSlicer/commit/3080289ab4d50eb36979cd8db0524e4e1fa7fd75

lukasmatena, 2022-05-19:
@gudvinr We tried to remove the override after upgrading to wxWidgets 3.1.6. It is true that the application no longer crashes. However, it is also practically unusable:
[image]

It may or may not have a simple fix. As of now, we are not investing time into it, there are higher priority issues. If someone knows how to fix it, let us know. Thanks.

When these issues are fixed (providing it is not caused by a bug in wxWidgets), we will disable the `GDK_BACKEND` override for wxWidgets 3.1.5, as shown below:
<details>
  <summary>Click to expand!</summary>
  
```diff
--- a/src/PrusaSlicer.cpp
+++ b/src/PrusaSlicer.cpp
@@ -57,6 +57,10 @@
     #include "slic3r/GUI/GUI_Init.hpp"
 #endif /* SLIC3R_GUI */

+#ifdef __WXGTK__
+    #include <wx/version.h>
+#endif
+
 using namespace Slic3r;

 static PrinterTechnology get_printer_technology(const DynamicConfig &config)
@@ -71,10 +75,13 @@ int CLI::run(int argc, char **argv)
     set_current_thread_name("slic3r_main");

 #ifdef __WXGTK__
+    #if !wxCHECK_VERSION(3, 1, 5)
     // On Linux, wxGTK has no support for Wayland, and the app crashes on
     // startup if gtk3 is used. This env var has to be set explicitly to
     // instruct the window manager to fall back to X server mode.
+    // This was supposedly fixed in wxWidgets 3.1.5 (https://github.com/prusa3d/PrusaSlicer/issues/8284)
     ::setenv("GDK_BACKEND", "x11", /* replace */ true);
+    #endif
 #endif

        // Switch boost::filesystem to utf8.

</details>

gudvinr, 2022-05-19:
@lukasmatena thanks for investigation. Do you see these issues only if Wayland backend is used or is it an issue you see in X11 too with wx 3.1.6?

lukasmatena, 2022-05-19:
@gudvinr As far as I know, the issue is on Wayland only. @tamasmeszaros will correct me if not.

mangado, 2022-07-04:
read this!
gtk4 reference - migrating from gtk3 to gtk4:
"Stop using grabs
GTK 4 no longer provides the gdk_device_grab() or gdk_seat_grab() apis"

Not only, there is no any notification when you move window.

Not only in GTK4!
In session wayland+gtk+3 also, no grab, no move notifications...

Codeblocks uses wxAui... and you can't move undocked widgets!
Probably thats (no grab, no move notification) are source of your issues...

bubnikv, 2022-07-04:
@tamasmeszaros investigated the Wayland / OpenGL context issue. It took him multiple days to produce a simple reproducible code snippet to reproduce the issue.
https://github.com/wxWidgets/wxWidgets/issues/22580

Until this issue is resolved, we will have to resort to X11 emulation.

gudvinr, 2022-07-11:
Thank you guys for doing a research and even going beyond that and creating high quality issue report. I hope now since wx released 3.2 now it will get some traction there.

mijoharas, 2022-07-17:
Hi, I'm also getting a segfault that appears to be at `wxTranslations::SetLanguage(wxLanguage)` (running prusa-slicer on wayland). Is there any further debugging help needed?

It sounds like you've got a repro and everything, but let me know if there's anything I can do to help.

Is there any workaround for this? I immediately get a segfault whether I run with GDK_BACKEND=x11 or not.

lukasmatena, 2022-07-18:
@mijoharas what build are you using and where you got it?

gudvinr, 2022-07-20:
I have this issue with arch build but I assume that is because they switched to 3.2 and this is exact same issue as we discussed before which happens on 3.1.6.  
So either PS should be hotfixed through patch in arch build system or via minor PS release to make it work or it'll just remain broken until 2.5.0 arrives in repo.

mijoharas, 2022-07-25:
Yep, I'm also using the arch build and having this problem.

lukasmatena, 2022-07-25:
@gudvinr @mijoharas Your issues have nothing to do with Wayland or PrusaSlicer. They are caused by Arch packaging: https://github.com/prusa3d/PrusaSlicer/issues/8299#issuecomment-1186748653

gudvinr, 2022-07-25:
Yeah, that is basically what we were talking about in may
> I believe that YuSanka has also seen the SetLanguage segfault when she attempted to build against wxWidgets 3.1.6, it is related to something that has changed in wxWidgets between 3.1.5 and 3.1.6. Not related.

gudvinr, 2023-09-06:
@lukasmatena issue was possibly fixed in https://github.com/wxWidgets/wxWidgets/commit/952de605f622cc7a3c30257365046ecd1a307808

tibordp, 2024-08-27:
I got the native Wayland build working by upgrading wxWidgets & some other small fixups in 
https://github.com/prusa3d/PrusaSlicer/pull/13307

I didn't rest it extensively, so it's probably subtly broken in some places, but the basic slicer functionality & webview all seem to be working fine (I'm on Fedora 40 with GNOME)

tibordp, 2024-08-28:
> > I got the native Wayland build working by upgrading wxWidgets & some other small fixups in
> > #13307
> 
> It is likely won't be accepted. You just replaced prusa's patched version of wxWidgets to vanilla wxWidgets without revisiting existing patches.
> 
> Either patches should be upstreamed or dropped if not needed anymore (and that's what prusa should do) or you need to update [prusa3d/wxWidgets](https://github.com/prusa3d/wxWidgets) instead.

That's fine, I don't really have the expertise to get the PR into the state where it could be merged as-is, as I cannot even test it on Windows os MacOS. It's more a proof-of-concept and if someone wants to do a private build that's mostly functional. 
 
I looked into the Prusa patches on top of wxWidgets 3.2.0 and it seems to be mostly about the dark mode functionality. If I have some time I'll try cherrypicking them on top of 3.2.5.

rom1dep, 2026-09-05:
In all appearances, PS is still using (sigh 😢) the same wxWidgets GUI library and the same rendering stack that defaults to X11. Everything above is still applicable, unfortunately.

eternal-sorrow, 2026-09-05:
>(sigh 😢)

What's wrong with wxWidgets?

rom1dep, 2026-09-05:
I'm not there to start a toolkit war, but let's say that:
- this issue is one and far from the only
- the whole PrusaSlicer UX rewrite was announced days apart from a [high-profile wxWidget project abandoning it](https://www.audacityteam.org/audacity-4/), and for the better