prusa3d/PrusaSlicer #8285: Slicing github skyline results in gcode error

ArchangelDesign, 2022-05-04:
### Description of the bug

This has already been reported as a feature request,
I just want to add how version 2.2 differs from 2.4

It appears you cannot slice my github skyline successfully on either version but you can fix the file in version 2.4 (through Netfabb) and then successfully slice in version 2.2 but you can't in version 2.4 because of the error in gcode, but I'm not importing gcode. :) 

feature request:
https://github.com/prusa3d/PrusaSlicer/issues/8093

### Project file & How to reproduce

1. Download my skyline in STL
2. Open in version 2.4
3. Object Settings -> Fix through Netfabb
4. Object settings ->export to STL
5. (now for reference open in version 2.2 which works fine)
6. Open the file again and you are still presented with an error and unable to slice

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.1

### Operating system

win64

### Printer model

ender 3v2

rtyr, 2022-05-04:
I am not completely sure what issue are you reporting. There is no screenshot or 3MF project file.

My guess is that you are getting error regarding missing `G92 E0`, which is likely caused by incorrectly configured printer profile. We started checking it since 2.4.1-beta1. PrusaSlicer just tells you to fix your printer profile.

As for the skyline model (not related to "G92 E0" error). I downloaded the model, repaired, and sliced it without issues. The model have errors, that is why it needs to be repaired. The issue with these skyline models was already reported here https://github.com/prusa3d/PrusaSlicer/issues/6159.

Example, how to fix the "G92 E0" error (if you are using relative extruder addressing):

[image]

From changelog:
 
> FDM printer firmware usually keeps an absolute extruder axis position using a single precision float representation, which leads to a loss of accuracy if the extruder axis is not resetted regularly. In extruder absolute mode 'G92 E0 is issued at each retraction by PrusaSlicer, while in extruder relative mode it is newly enforced, that either layer_gcode or before_layer_gcode contains 'G92 E0'. On the other side, in extruder absolute mode, 'G92 E0' in layer change custom G-code breaks slicer's assumption leading to horrible over-extrusions, thus 'G92 E0' is newly tested for and prohibited with extruder absolute addressing https://github.com/prusa3d/PrusaSlicer/issues/6336 https://github.com/prusa3d/PrusaSlicer/issues/5073.

rtyr, 2022-05-05:
Can we close the issue?

closed 2022-05-05 by ArchangelDesign