prusa3d/PrusaSlicer #8286: Slicing button is gone

ivanhandler, 2022-05-04:
### Description of the bug

When I load a new object (stl) to be sliced the image remains green and the button that was for slicing is replace with an export to gcode is there instead.  It actually does the slicing instantaneously - probably because I have a simple model.  But now I can't see what the final looks like until after I print it.  Have the latest version 2.4.2.
[image]

### Project file & How to reproduce

[Excavated_dodecahedron.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/8627897/Excavated_dodecahedron.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

macOS Monterey 12.3.1 - Mac mini

### Printer model

SeeMeCNC Boss 300

jmd73, 2022-05-05:
@ivanhandler you probably have automatic slicing activated in preferences. 
Regarding the layer view, you can switch between model view and layer view with the widgets in the lower left corner of the screen (see your own image)

lukasmatena, 2022-05-05:
Exactly as @jmd73 says. You apparently have 'Background processing' activated in Preferences, which means that you yourself decide when to switch between the 3D scene and G-Code preview (in the bottom left corner). When it is deactivated, PrusaSlicer switches to G-Code preview automatically when you press the "Slice Now" button (which would show in that situation).

Can we close the issue?