prusa3d/PrusaSlicer #8288: Unexpected offset motion of print head, during print

DrJuJu, 2022-05-05:
### Description of the bug

Since upgrading to 2.4.0 / 2.4.2, while printing, on each layer, the head moves outside of the print area, pauses, then returns to the part being printed.  

This has only started happening on multiple printers with gcode files produced on multiple computers, and since upgrading to 2.4.0/2.4.2. It appears to be acting like the head is moving to a wipe tower, but no filament is extruded.
[Arducam_IMX519_Case_v3_0.15mm_ABS.gcode.txt](https://github.com/prusa3d/PrusaSlicer/files/8630036/Arducam_IMX519_Case_v3_0.15mm_ABS.gcode.txt)
[image]
[20220505T093003Z.zip](https://github.com/prusa3d/PrusaSlicer/files/8630078/20220505T093003Z.zip)
[PrusaSlicer.ini.txt](https://github.com/prusa3d/PrusaSlicer/files/8630088/PrusaSlicer.ini.txt)

### Project file & How to reproduce

[20220505T093003Z.zip](https://github.com/prusa3d/PrusaSlicer/files/8630095/20220505T093003Z.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.0 & 2.4.2

### Operating system

Windows 10

### Printer model

Prusa i3 Mk2 & Voron 2.4

lukasmatena, 2022-05-05:
The G-Code Viewer does not show any suspicious travels as far as I can see. Can you please
1. Provide a 3MF file
2. Show the exact spot in the G-Code above where the extra travel occurs
?

Thanks.

DrJuJu, 2022-05-05:
Sorry I don't normally use 3mf files, so I've had to go through an online conversion site...
[fabconvert.com_arducam_imx519_back_case_v3.3mf.txt](https://github.com/prusa3d/PrusaSlicer/files/8630284/fabconvert.com_arducam_imx519_back_case_v3.3mf.txt)
[fabconvert.com_arducam_imx519_front_case_v1a.3mf.txt](https://github.com/prusa3d/PrusaSlicer/files/8630285/fabconvert.com_arducam_imx519_front_case_v1a.3mf.txt)
.
As for a specific point in the gcode files, like you I see nothing obvious when using a gcode viewer, so I'm wondering if it's related to a non movement config setting ??

rtyr, 2022-05-05:
> Sorry I don't normally use 3mf files

What Lukas meant is our standard 3MF project file, which can be easily exported using File - Save project as...

As for your issue:

- The behaviour you are describing is usually caused by some sort of communication error (SDcard issue, cardreader issue, LCD wiring issue, Octoprint communication issue, ...).

- I am printing your g-code on stock i3 MK2S, no issues

rtyr, 2022-05-05:
It can be also just some Octoprint plugin, like Octolapse.

DrJuJu, 2022-05-05:
Here's the 3mf project file, As for the issue, if I was only seeing it on the Octoprint/Prusa system, then I'd agree with you, but I'm also seeing it on a Pi/Duet3d/Voron2.4 system.

[Arducam_IMX519_Case_v3.3mf.txt](https://github.com/prusa3d/PrusaSlicer/files/8630486/Arducam_IMX519_Case_v3.3mf.txt)

rtyr, 2022-05-05:
I just printed your gcode without problem on MK2S.

DrJuJu, 2022-05-05:
Thanks for the update, I'll have to check everything again....

rtyr, 2022-05-05:
I would start by printing it directly from SDcard.

closed 2022-05-09 by stepikovo