prusa3d/PrusaSlicer #8292: 2.4.1 and 2.4.2 - Can't open any STL or 3MF file on iMac

Jiheffe, 2022-05-06:
### Description of the bug

Problems started when I upgraded to 2.4.1. Whenever I try to open an STL of 3MF file, I get this message "Prusaslicer has encountered an error - Loading of a model failed".
[image]

I somehow managed to get it working again, but I do not remember how. I just know I went through uninstalling and reinstalling the whole thing, which lots of trouble to find all my saved prests...
Now just went from 2.4.1 to 2.4.2 and here we go again 😟  I can't load any file into PrusaSlicer...

### Project file & How to reproduce

[Bug Report.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/8641274/Bug.Report.3mf.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.4.2+x64

### Operating system

MacOS Monterey 12.3.1 on iMac Retina 5K, 27-inch, Late 2015, Intel based.

### Printer model

Prusa i3 Mk3S + MMU2S

Jiheffe, 2022-05-06:
Problem solved. My files are on Dropbox, and are removed locally when not used for some time.
Before I can open them in PrusaSlicer, I need to manually make them available offline. This seems to be a change in Dropbox behaviour.

closed 2022-05-06 by Jiheffe

sslupsky, 2022-05-21:
@Jiheffe I encountered this issue with iCloud recently as well.  From what I understand, if the file is not available, the OS notifies the application and the application is supposed to request that the file download.  Perhaps this issue needs to be reopened?