prusa3d/PrusaSlicer #8293: what if i dont want to save a file and shut off the program?

CaVO32, 2022-05-07:
### Description of the bug

if i use this slicer it wont turn off unless i save a copy of a file somewhere on my hdd....
if i open  the app and do not load any item onto the build plate then i can shut off the app as long as i do not change anything within the app such as overviewing a profile or looking at settings ect...
but if i load any item onto the build area and even if i delete it after viewing only without changing anything it still obligates me to save a file to my hdd before it will shut down.
what if i dont want to save the file? what if it was just a test? how about allowing to shut off the program without obligating to put something somewhere ?

### Project file & How to reproduce

1 on a mac... open prusa slicer app
2 load any stl file......
3 try as you may to shut off the app without saving the file
4 after getting frustrated   ... delete the file from the build plate and try to shut down the app without saving a empty file 
 5 open activity monitor for mac and find prusa slicer and force quit the app
6 go to github and write this because it is a pain 
7 wait for a reply

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

latest

### Operating system

macos

### Printer model

plastispitter f40m t1ni hole

neophyl, 2022-05-07:
You say you are using the latest version ?  But which version exactly. As this issue sounds like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/8224 

That issue mentions it’s fixed in 2.4.2

lukasmatena, 2023-02-06:
No response, closing. If a duplicate of #8224, it is fixed for a long time.

closed 2023-02-06 by lukasmatena