prusa3d/PrusaSlicer #8294: Inconsistent number of layers produced for same part height depending on overall model height

edwardhorsford, 2022-05-07:
### Description of the bug

I've got a part that consists of a base plate (1.9mm), and some raised text above it. The raised text varies in height between models.

1.9mm is between layer heights, so PrusaSlicer is having to pick how many layers to do it with. What I'm observing is that if my text height is varied, the base section unexpectedly slices to a different number of layers.

I've set up PrusaSlicer to change colour as the text changes.

See screenshot:
[image]

Both parts have the same 1.9mm thickness. Yet the upper part is all orange and using 9 layers. The lower part has one layer white, and is 10 layers for the same thickness.

The difference between the STLs is that the text `BOSCH` is 0.7mm higher than other text, whilst `Makita` is the same height as the rest of the text.

---

NB: I appreciate these model heights aren't ideal - this was a downloaded model and I'll probably adjust them to be on 0.2mm boundaries. But it's unexpected that the overall model height or would cause PrusaSlicer to choose a different number of layers for something *below*. I'd assume that 1.9mm *always* produces the same number of layers - but here that doesn't seem to be the case.

### Project file & How to reproduce

[L-Boxx labels.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/8644912/L-Boxx.labels.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

MacOS 11.6

### Printer model

Prusa Mini+

foreachthing, 2022-05-07:
Doesn't this look correct?
I opened it in 2.5.0a0. No changes.
[image]

edwardhorsford, 2022-05-07:
@foreachthing yes it does.

Curiously I've just opened the same project file in a fresh PrusaSlicer instance and it looks ok too. I wonder if something is different between dragging STL's in and slicing vs opening from a project file.

edwardhorsford, 2022-05-07:
I can reproduce by dragging in the STLs, but not from the project file. Attaching a zip of the STLs. If you add them both and slice, you'll see that the base plate of one has an extra layer.

Screenshot showing that text is drawn on one a layer earlier than the other:
[image]

[Model STLs.zip](https://github.com/prusa3d/PrusaSlicer/files/8645189/Model.STLs.zip)

edwardhorsford, 2022-05-07:
Uploading a project file for the above which *does* seem to show the issue. Unsure why the first project file doesn't.
[L-Boxx labels revised.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/8645190/L-Boxx.labels.revised.3mf.zip)

closed 2026-09-29 by prusaslicer-bot[bot]