prusa3d/PrusaSlicer #8298: Overlap options

mariusimv, 2022-05-08:
**Is your feature request related to a problem? Please describe.**
The Infill/perimeters overlap is a great idea, but sometimes you might need to set this overlap at layer/function level. As is now, the overlap applies to all layers. It is great for internal part infill but no so great for the first layer, top layer or solid infill.
If we use the overlap, the first layer will have too much material at the interface with the perimeters. It also applies for the solid infill.

**Describe the solution you'd like**
It would be great if this setting can have different values or flowrates for: first layer, solid infill, top layer and internal infill.
[image]

kubispe1, 2022-09-19:
Duplicate issue: #4629

ronperek, 2024-03-06:
+1 for this feature request
I have a print that have large solid infill area (10x10cm). If overlap is set to more than 0% then I can clearly see that near perimeters there is over extrusion. This over extrusion builds up on the nozzle and is released later as a filament blob somewhere in the print.
The extrusion multiplier is calibrated. Perimeter with is correct. Also, in the middle of solid infill area, there is no over extrusion.
I don't want to set overlap to 0%, because it's good for part strength when the infill is not 100%.
Guys at SuperSlicer already added this feature:
https://github.com/supermerill/SuperSlicer/issues/2028

closed 2026-09-29 by prusaslicer-bot[bot]