prusa3d/PrusaSlicer #8301: retract_lift should be disabled when FW retraction is enabled

gudvinr, 2022-05-08:
### Description of the bug

In printer settings, when you enable `use_firmware_retraction`, most of the retraction options are disabled except `retract_lift` which is always enabled.

[M207 G-Code](https://www.reprap.org/wiki/G-code#M207:_Set_retract_length) has configuration for z-hop, which is essentially the same thing. It is supported by Prusa, Marlin and RepRap.

There are two issues with disabling those, however:
* `retract_lift_above` and `retract_lift_below` can't be used too. If that is needed though, this can be tuned with custom G-Code and conditional `M207 Z` G-Code
* klipper currently doesn't support z-hop setting. That requires either retraction G-Code to be flavor-dependent or custom macros for klipper

### Project file & How to reproduce

No need for project files since it doesn't affect one particular project.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Arch Linux, KDE Plasma 5.24

### Printer model

Artillery Genius Pro, Marlin 2.0.9.3

SchmartMaker, 2022-05-09:
I'm also using firmware retraction with RepRapFirmware. Disabling `retract_lift` might make the UX consistent at least. But for my use case it would be better if none of the retraction-related fields were disabled, since I'm using the values from the user interface to configure `M207` in the `Start G-Code` script as follows:

```plain
M207 S[retract_length] R[retract_restart_extra] F{retract_speed[current_extruder]*60} T{deretract_speed[current_extruder]*60} Z[retract_lift]
```

Fortunately, as a workaround, the retraction fields in the `Filament Overrides` let me specify the values on a per-filament basis, but having these fields enabled to also be able to define the settings on the **machine-level** would be welcome. But perhaps my way of doing this is weird and I've missed a more elegant method.

gudvinr, 2022-05-10:
> I'm using the values from the user interface to configure `M207` in the `Start G-Code`

That is so clever! I actually agree with that so I'll probably edit this issue instead.

closed 2026-09-29 by prusaslicer-bot[bot]