prusa3d/PrusaSlicer #8302: boost-1.79.0 fails even with latest master.

slashbeast, 2022-05-08:
### Description of the bug

The 2.4.2 as well as current master branch does not build with boost-1.79.0.

Relevant errors:

    /home/piotr/src/PrusaSlicer/src/slic3r/Utils/AppUpdater.cpp:245:52: error: ‘fstream’ is not a member of ‘boost::filesystem’; did you mean ‘std::fstream’?
    /home/piotr/src/PrusaSlicer/src/slic3r/Utils/AppUpdater.cpp:246:33: error: ‘file’ was not declared in this scope
    /home/piotr/src/PrusaSlicer/src/slic3r/GUI/DesktopIntegrationDialog.cpp:189:30: error: variable ‘std::ofstream output’ has initializer but incomplete type
    /home/piotr/src/PrusaSlicer/src/slic3r/GUI/GUI_App.cpp:865:30: error: variable ‘std::ifstream stream’ has initializer but incomplete type

### Project file & How to reproduce

- Update boost to 1.79.0
- Fetch sources
- mkdir build && cd build
- cmake .. -DOPENVDB_FIND_MODULE_PATH="/usr/lib64/cmake/OpenVDB" -DSLIC3R_FHS=ON -DSLIC3R_GTK=3 -DSLIC3R_GUI=ON -DSLIC3R_PCH=OFF -DSLIC3R_STATIC=OFF -DSLIC3R_WX_STABLE=ON -Wno-dev
- make -j$(nproc)
- see errors.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2, master

### Operating system

Linux, Gentoo, ~amd64

### Printer model

mk3s+

slashbeast, 2022-05-09:
Pull request with needed fixes is up. https://github.com/prusa3d/PrusaSlicer/pull/8304

thesamesam, 2022-05-10:
See also: #8238.

PR #8304 (2022-05-09): Fix building with new Boost 1.79.0.
Fixes #8302
commit 436a454b (2022-05-10): Fix building with new Boost 1.79.0.

Fixes #8302