prusa3d/PrusaSlicer #8303: Filament vendor list needs better management

timur-tabi, 2022-05-09:
Prusaslicer needs to rethink the way it handles filament vendors.  The current method is unwieldy.  There are a few problems:

1) A lot of vendors, or at least a lot of filament types from some vendors, are just missing.  There's no Hatchbox PETG.  There's no Polymaker PLA.

2) It's unrealistic to expect the user to select which vendors/filaments he wants when installing Prusaslicer.  I suspect most users don't consistently use the same vendor year after year.

3) If you don't choose a vendor during installation, there doesn't appear to be a way to add a new vendor to the list.

4) The selection drop-down doesn't let you filter by vendor.  This is cumbersome:
[image]

5) The cost is hard-coded in the filament profile, which makes no sense.  The cost varies significantly from day to day and person to person.  Worse yet, if you change the cost, you're forced to create a new profile, which just makes things even messier.

We need a better way to manage and select filaments.  Some ideas:

1) All profiles should be installed by default.  On the Filament Settings page, there should be a drop-down to select the vendors that the user wants.  So if the user one day decides to try a new vendor, all he needs to do is add it from the list.

2) "Diameter" needs to be an Expert setting.  Seriously, does anyone even make a printer that can support more than one diameter?  And no one is going to measure their filament spool to determine that the average diameter is actually 1.74mm instead of 1.75mm.

3) The filament drop-down needs to be divided into categories, with the type of filament (PLA, PETG, etc) being the first level.  Then after that is the vendor, and then the third level is the type of filament.

4) We need a way to assign profile values to specific colors.  Hatchbox Wood PLA prints very differently from Hatchbox Black PLA.  I don't want to have to create entire new profiles for different colors.

5) It should be possible to change the cost of the filament without creating a new profile.

I'm sure there are more ideas but this is a good start, IMHO.

neophyl, 2022-05-09:
On your point 2, yes people do measure their filament diameter and accurize the value as it’s one of the most important steps in tuning your filament settings. I do it will all my filaments. 
Then again I save custom copies for all the ones I use even if there is a system version. This allows me to alter the price and any other value over time and save it. That way I don’t have problems with trying to over write one of them which isn’t allowed.

rtyr, 2022-05-09:
> A lot of vendors, or at least a lot of filament types from some vendors, are just missing. There's no Hatchbox PETG. There's no Polymaker PLA.

Are you saying we must always add the whole range of filaments for each vendor? It is mainly about filaments, not about vendors. You are buying specific filaments, right? Should we always wait till we have profiles for every single filament manufactured by some vendor? I don´t understand this logic. 

> It's unrealistic to expect the user to select which vendors/filaments he wants when installing Prusaslicer. I suspect most users don't consistently use the same vendor year after year.

And that is the reason we implemented filament wizard, so you can add/remove filament profiles at any time, not just when installing PrusaSlicer.

> If you don't choose a vendor during installation, there doesn't appear to be a way to add a new vendor to the list.

That is not correct. You can enable/disable filament profile at any time, simply by going to Configuration -> Configuration wizard or just by using "Add/Remove filaments button" at the end of filament selection...

[image]

> The cost is hard-coded in the filament profile, which makes no sense. The cost varies significantly from day to day and person to person. Worse yet, if you change the cost, you're forced to create a new profile, which just makes things even messier.

This will be adressed in PS2.5.0.

> All profiles should be installed by default. 

I don't think so. The whole point of filament wizard is to not install all unnecessary filament profiles. There is no reason for it, since you can enable/disable filament profiles at any time.

julie777, 2023-01-27:
I would expect that a tree would be a better view to show all known filaments.
- vendor
- filament type
- diameter
- color

I should be able to select the ones that I use and put them in a separate list, which is the list in the dropdown when viewing the plate.

Cost

I do believe that cost should be stored as part of the filament definition. However, cost can fluctuate so it would be nice if the cost was shown next to the filament choice in the right sidebar, making always visible so I know what the sliced cost of the part is based on. It should allow changing it which would update the filament definition. That way it would be visible and convenient.

timur-tabi, 2026-09-02:
This is still a useful feature to have.