prusa3d/PrusaSlicer #8306: Perimeter overlap limiting

willkaxxx, 2022-05-09:
* Here the perimeters overlap too much, resulting in a defect (see below)
[image]

This wall shold be straight
[image]

* I suggest to add settings to limit maximum perimeter overlap and replace this two perimeters with one solid infill

Somthing like this:
[image]

Simmilar behavior is in `Simplify3D` slicer

willkaxxx, 2022-06-12:
Jest enable thin wall detection...)

closed 2022-06-12 by willkaxxx

NGC869, 2022-06-29:
> Just enable thin wall detection...)

**No, this is not a true solution!**
The "Thin wall detection" feature is able to avoid the exposed problem but introduce other problems for other kinds of prints (for example look at: https://github.com/prusa3d/PrusaSlicer/issues/7973 )

PrusaSlicer deserve to be enriched with a more intelligent way to recognize thin walls and print them in the most wise way.

lukasmatena, 2022-06-29:
@NGC869 Can you please try this in PrusaSlicer 2.5.0-alpha2, which was just released? https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.5.0-alpha2