prusa3d/PrusaSlicer #8307: Export toolpaths to OBJ - Render mode

The3Designer, 2022-05-10:
**Is your feature request related to a problem? Please describe.**
I use the "export toolpaths to OBJ" a lot to create (somewhat) realistic renders of 3D printed parts. However, this option exports **all** toolpaths, including the skirt and prime line. The skirt can be disabled quite easily, but removing the prime line from the G-code is a bit more cumbersome. I currently remove the prime line from within the rendering software, but this becomes problematic when I update the model (i.e. reload the OBJ file after a design change), since this reload will also re-import the prime line.

**Describe the solution you'd like**
For rendering, I only need the sliced object, not the "peripheral" G-code. Therefore, it would be nice to have a sub-function or seperate function that is based on the "export toolpaths to OBJ", but that only exports the toolpaths of the sliced model. This function would strip things like skirt, prime line, draft shield, etc. from the exported .OBJ file, without these toolpaths being removed from the actual G-code.

**Describe how it would work**
How this functionality would be implemented in the back-end is unknown to me. For the GUI, a second function (button) could be included in the "File --> Export" menu, named: "Export Object Toolpaths as OBJ...". To improve the clarity between this new function and the existing "Export Toolpaths as OBJ..." function, the latter could be renamed to "Export All Toolpaths as OBJ...".

**Describe alternatives you've considered**
I currently remove the skirt in the printing settings in PrusaSlicer, and then remove the prime line from within the rendering software. Not an ideal workflow (as explained in the first section).

**Additional context**
n.a.

enricoturri1966, 2022-05-10:
@The3Designer
The 'Export Toolpaths to OBJ...' command saves into the obj file the current content of the preview.
This means that you can turn off the visibility of skirt and/or prime lines (in the legend) and then export the remaining toolpaths.
The same applies if you want to export only a subset of layers/toolpaths (try playing with the two sliders and export)

The3Designer, 2022-05-10:
> @The3Designer The 'Export Toolpaths to OBJ...' command saves into the obj file the current content of the preview. This means that you can turn off the visibility of skirt and/or prime lines (in the [...]

Oh wow, I didn't know that functionality worked like that. Your steps of hiding the unwanted toolpaths worked perfectly!

Wish I knew this before haha, works like a charm ;)
I'll close this issue, since this is an okay workflow IMO.

closed 2022-05-10 by The3Designer