prusa3d/PrusaSlicer #8311: Mac File Operations Always Requests Access to Documents Folder

njjerrysmith, 2022-05-10:
### Description of the bug

After every time I boot the software, on the first time I go to open a file from the recent projects menu, the system dialogue ""PursaSlicer" would like to access files in your Documents folder" pops up despite access being granted in the system preferences "file and folders" privacy pane.  

[image]

[image]

PrusaSlicer
Version:   2.4.2+x64
Build:     PrusaSlicer-2.4.2+x64-202204251114

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Version 12.3.1 (Build 21E258)
Total RAM size [MB]: 17,180MB
OpenGL installation
GL version:   2.1 ATI-4.8.15
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon Pro 450 OpenGL Engine
GLSL version: 1.20
<details>
<summary>Installed extensions:</summary>
GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_shadow_ambient
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_blend_weighted_minmax
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_depth_bounds_test
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SGI_color_matrix
</details>

### Project file & How to reproduce

[Archive.zip](https://github.com/prusa3d/PrusaSlicer/files/8665576/Archive.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2+x64

### Operating system

macOS 12.3.1

### Printer model

i3 MK3S

lukasmatena, 2022-05-13:
@njjerrysmith Thanks. Do you have a single PrusaSlicer installation in Apps, or do you also run other builds that you downloaded elsewhere (such as our github builds)? I am frankly not sure about how exactly it works in macOS.

njjerrysmith, 2022-05-13:
No problem. I downloaded the universal binary from the github releases page and dragged it into /Applications/Original Prusa Drivers. It replaced the old stable release. I have run some beta builds that were installed to /Applications, but currently don't have any installed and didn't have any installed when I updated to 2.4.2

I think I remember allowing file access to downloads and the desktop as well, but I only did that on the first boot, and those prompts haven't come back. Though they aren't listed in the system preferences pane like they are for other software (see raspberry pi imager), so maybe I'm miss-remembering.

fnsign, 2022-05-19:
I'm having the exact same issue running on macOS Catalina 10.15.7 since I updated to PS 2.4.2 from the Github package.
No other builds ever installed.

dhysuiej, 2022-11-06:
Any ETA on this issue. I'm having the same problem with PrusaSlicer 2.5.0 on MacOS 10.15.7 (Catalina).

nknotts, 2022-12-07:
I am having the same problem.

I am running macOS 12.6.1 (Monterey - M2 MacBookAir) 
Version 2.5.0+MacOS-arm64

From google, I've found other applications that have similar issues. It seems to be code signature related
https://github.com/alacritty/alacritty/issues/5840

After running
`sudo codesign --force --deep --sign - /Applications/PrusaSlicer.app`
The permission modal stopper appearing after the first prompt.

ulfertg, 2023-02-10:
Same problem here.

MacBook Pro M1 Max, macOS Ventura 13.2
PrusaSlicer Version 2.6.0-alpha3+MacOS-arm64 (but also on all previous versions I've been using, starting with 2.4.2)

I tried signing the app with the command mentioned above, but received this error msg: "Operation not permitted"

nknotts, 2023-02-10:
@ulfertg Try https://osxdaily.com/2018/10/09/fix-operation-not-permitted-terminal-error-macos/

ulfertg, 2023-02-11:
@nknotts Thank you! Finally the pop ups stopped.

ajaytanna, 2023-11-22:
> I am having the same problem.
> 
> I am running macOS 12.6.1 (Monterey - M2 MacBookAir) Version 2.5.0+MacOS-arm64
> 
> From google, I've found other applications that have similar issues. It seems to be code signature related [alacritty/alacritty#5840](https://github.com/alacritty/alacritty/issues/5840)
> 
> After running `sudo codesign --force --deep --sign - /Applications/PrusaSlicer.app` The permission modal stopper appearing after the first prompt.

This needs to be done even in the latest PrusaSlicer 2.7.0 RC2 Same popup about access to Documents folder every time progam is launched.

closed 2026-09-29 by prusaslicer-bot[bot]