prusa3d/PrusaSlicer #8312: Can support Firmware Klipper ?

Witawat, 2022-05-11:
Can support Firmware Klipper ? 
[image]

rtyr, 2022-05-11:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3452. We will likely add separate Klipper gcode flavor, but you can use Marlin legacy flavor for printer with Klipper FW (as recommended by Klipper author).

I am closing this as a duplicate.

closed 2022-05-11 by rtyr

shawe, 2023-03-19:
> Duplicate of #3452. We will likely add separate Klipper gcode flavor, but you can use Marlin legacy flavor for printer with Klipper FW (as recommended by Klipper author).
> 
> I am closing this as a duplicate.

We can do this, but only if you don't add some extra-code automatically, asuming that we are missing some thing.

With macros on Klipper that don't exists on Marlin, we can simplify code on slicer, customizing it from printer configured macros.

But if you consider that ower start_gcode must have a M140, M104, M190 or M109 and auto-added for me, you are adding something unwanted. Or like in this case, forcing me to add duplicated gcode because your slicer be sure that I'm not missing something important.

For me, this two start_gcode must do the same from PrusaSlicer, and you can verify that result is diferent:

```
START_PRINT EXTRUDER_TEMP={first_layer_temperature[0]} BED_TEMP={first_layer_bed_temperature[0]}
```

```
; Solve PrusaSlicer BUG adding temperatures that it wants
M140 S45   ; Set Bed Temperature
M104 S160  ; Set Hotend Temperature
M190 S45   ; Wait for Bed Temperature
M109 S160  ; Wait for Hotend Temperature

START_PRINT EXTRUDER_TEMP={first_layer_temperature[0]} BED_TEMP={first_layer_bed_temperature[0]}
```

If you try to print ABS with bed to 90, the first start_gcode will be modified by PrusaSlicer to auto-add a prehead of bed to 90, when the second was to explicitly added.

In both cases, my macro was yet controlling that, and result unnecesary if you don't add something unwanted. But because PrusaSlicer decided to add my bed was doing:

1. Pre-head to 90
2. Wait to 90 to be reached
3. Pre-head to 45
4. Wait to 45 to be reached

More power and time consuming that really was needed, only because you think that I miss something.

Please only put the gcode that is really needed, one start_gcode or other, but not a third attemp trying to add something than can be missed without at least a warning.