prusa3d/PrusaSlicer #8313: A wish about customise support structure

JSssssss, 2022-05-11:
**my mighty developer**
I'm now working on a copy of logitech m705

[image]

**I'd like to create support myself as parts in save of time**
is that possible to let me define which part is cheap support and which is expensive frame outline？

**these thin walls saved a ton but still take 2 path to finish**
i have to set the wall to 0.46mm minimum to keep them sliced or they won‘t be considered as a part under .28mm superdraft preset
maybe make the wall thinner than .46mm auto changed to support structure will be a cool feature

[image]

**thank you very much for reading this**
and this is time cost compare
[image]

stepikovo, 2022-05-15:
Unfortunately, you have not provided the file, but can you try _Detect thin walls_ if that solves the issue?

JSssssss, 2022-05-16:
> Unfortunately, you have not provided the file, but can you try _Detect thin walls_ if that solves the issue?

im sorry i don't make myself clear, there is nothing to be solved this is just a functionality request
i just want to turn model feature into support structure
i edited the post see if makes any batter

rtyr, 2022-05-17:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3390.

closed 2022-05-17 by stepikovo