prusa3d/PrusaSlicer #8318: Scale to proportion

kkolodziejczak, 2022-05-11:
**Is your feature request related to a problem? Please describe.**
I have downloaded a model for 16mm magnet, but I have a 17mm magnet at home.

**Describe the solution you'd like**
Have a button to scale to proportion, give current value, and desired.

**Describe how it would work**
Scale of the model should be set to bigger or smaller depending on the calculation.

**Describe alternatives you've considered**

**Additional context**
My current solution is to calculate this on the side, and then input it in the slicer. Option from the menu would be nice.

lukasmatena, 2022-05-12:
This is related to #3651 (measurement tool), in fact it is a duplicate. The "scale so this dimension is xyz" feature should be implemented as part of the measurement tool.

kkolodziejczak, 2022-05-12:
@lukasmatena I agree and disagree at the same time. That feature requests you have mentioned is huge, it would take multiple sprints to finish I guess. What I have in mind could be as a thing to time being, I guess it is small enough that it could be done in a single sprint.

lukasmatena, 2022-05-13:
@kkolodziejczak It may be easier to do, but also more complicated from the UI point of view to make clear what it does. And it would still require time to implement and test. I'd rather spend the time on the final and more universal solution, than on something that will be obsolete maybe even before the next big release. The measurement tool is now near the top of the priority list, it should be implemented relatively soon.

closed 2022-05-13 by kkolodziejczak