prusa3d/PrusaSlicer #8319: Autorotation Crashes App

pressionllc, 2022-05-11:
### Description of the bug

When attempting to rotate a part in Version 2.4.2+win64 using any of the auto orientation selections the app crashes. This crash was able to be duplicated by Prusa support.

### Project file & How to reproduce

[SirayaTest.zip](https://github.com/prusa3d/PrusaSlicer/files/8673794/SirayaTest.zip)

To recreate the problem add the 3MF file in the attached. Select one part, select rotate, pick "Reduce Overhand Slopes".

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.4.2+win64

### Operating system

Win 11

### Printer model

SL1S

lukasmatena, 2022-05-12:
Thanks, reproduced both in 2.4.2 and current master. I enclose a simplified 3MF which should be easier to debug: 
[SirayaTest_simple.zip](https://github.com/prusa3d/PrusaSlicer/files/8675569/SirayaTest_simple.zip). I had to open the 3MF, optimize for Best surface quality, then optimize for Reduced overhang slopes.

We'll look into it.

bubnikv, 2022-06-29:
Fixed in PrusaSlicer 2.5.0-alpha2. Closing.

closed 2022-06-29 by bubnikv