prusa3d/PrusaSlicer #8320: Favorite settings page

ps-igel, 2022-05-12:
**Is your feature request related to a problem? Please describe.**
Not related to a problem.

**Describe the solution you'd like**
I usually only change a few settings on my prints. I'd like to mark a setting as Favorite and then have a favorites page with exactly these settings. For example:

I usually change: 
- Infill density -> Infill page
- Number of perimters -> layers and perimeters page
- Loops -> Skirt and brim page
- Support style -> support material page
- Complete individual objects -> output options
- a few more

Right know I have to go through five different settings pages. I would like to have my favorite settings on one favorite settings page, see images below.

**Describe how it would work**
I could imagine a "set as favorite" toggle icon next to each option. Then all favorites that are toggled on will be displayed on the favorite settings page.

**Describe alternatives you've considered**
I could save each setting as a different print profile, but I would end up with a huge variety of profiles.

**Additional context**
see below

[image]

[image]

foreachthing, 2022-05-12:
Original Slic3r has that! Wish PS has it too!

dartrax, 2022-07-09:
Another possible solution: #1267

ulab, 2023-03-12:
> Another possible solution: #1267

Once you can mark settings as favourite, there could be a section of these faved settings in the sidebar. I love that idea.

lukasmatena, 2026-09-01:
Implemented in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena