prusa3d/PrusaSlicer #8321: 2.4.2 can't be opened, error -47

jonathanalevi, 2022-05-12:
### Description of the bug

I can't open the app, I get an error on MacOS with Apple M1X.. Error -47 

### Project file & How to reproduce

I just try to open the app, and it doesn't work. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2 ARM64

### Operating system

macOS Monterey 12.3.1

### Printer model

MK3S+

lukasmatena, 2022-05-13:
@jonathanalevi Are you definitely using the correct binary? You should be using https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.4.2/PrusaSlicer-2.4.2+universal-202204251114.dmg, not the ARM64 build.

jonathanalevi, 2022-05-17:
Strange, I'd downloaded the "Universal" and it didn't work, now, with your link, it works. Note, though, that the reason I thought it was arm64 was because of the splash screen

BTW, this version loads INSANELLLLY fast, so I think maybe before I wasn't running the right version for my processor lol.

closed 2022-05-17 by jonathanalevi

stepheneb, 2022-07-09:
I just experienced the same error -47 trying to open 2.4.2. Turns out I had v2.4.0 already open. After closing v2.4.0 I was able to open the newly downloaded universal build 2.4.2