prusa3d/PrusaSlicer #8324: Wrong temperaturs on MMU print

wbphelps, 2022-05-13:
### Description of the bug

Extruder 2 is set to Nylon with nozzle temps of (285,290).  Extruder 3 is PVA with nozzle temps of (210,205).  BUt when I slice and look at the temperature display it is exactly opposite, it shows support being printed at the higher temperature and the object at the lower temperature. Also tried displaying the g-code with g-code viewer and it shows the same. Tool display shows the correct extruders. 

### Project file & How to reproduce

[door handle test 9Cae PA6-CF PVA.zip](https://github.com/prusa3d/PrusaSlicer/files/8683999/door.handle.test.9Cae.PA6-CF.PVA.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2+win64

### Operating system

Windows 10 pro

### Printer model

original I3 MK3S+ with MMU2S

rtyr, 2022-05-13:
It seems the issue is only in the preview, the g-code looks ok.

cmwoodman, 2022-12-22:
Can confirm this occurs on the MacOS version as well.

closed 2026-09-29 by prusaslicer-bot[bot]