prusa3d/PrusaSlicer #8328: laden einer Konfiguration Fehler auf Mac

Darkmoon1000, 2022-05-14:
### Description of the bug

[image]
[image]
[image]
[image]

### Project file & How to reproduce

beim Laden der Konfiguration`s  Datei kommt diese Fehlermeldung ! bekomme diese nicht geladen.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Prusa 2.4.2+arm64

### Operating system

neuste MacOs Version

### Printer model

Prusa Mk3S+

stepikovo, 2022-05-16:
Hello @Darkmoon1000, can you please include the config file?

Darkmoon1000, 2022-05-16:
[Mac_Config_P_Slicer.zip](https://github.com/prusa3d/PrusaSlicer/files/8699857/Mac_Config_P_Slicer.zip)

Darkmoon1000, 2022-05-16:
sind beide ini dateien enthalten

stepikovo, 2022-05-16:
You have "**extrudr PETG glowEx**" twice in the config, there can't be duplicates, please rename one of them

closed 2022-05-16 by stepikovo

Darkmoon1000, 2022-05-16:
Vielen Dank , hat funktioniert