prusa3d/PrusaSlicer #8331: Paint on Support not working on detailed texture geometry

natehansen66, 2022-05-14:
### Description of the bug

Attempting to use paint on support blocker to keep supports off the texture on this part. This does not work at all on the detailed texture but works fine on the untextured areas of the model. I need to completely block supports from the texturing.
[image]
 Using the latest stable release.

### Project file & How to reproduce

[project.zip](https://github.com/prusa3d/PrusaSlicer/files/8693628/project.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Windows 10

### Printer model

MK3S

FidelCapo, 2022-05-16:
The project you attached does not include a model. Can you please upload the project with the model again? Thank you

stepikovo, 2022-05-16:
Can you send the model to info@prusa3d.com or that's also not an option?

stepikovo, 2022-05-16:
"Slicer issue - for Stepan" will be perfect, thanks.

natehansen66, 2022-05-16:
Email sent

stepikovo, 2022-05-17:
Thank you for the model, turns out the model is extremely detailed (over 1,3 M triangles and most of them on the grip) and the paint tool struggles to grab the facets. Until we try fixing/improving it, a workaround would be simplifying the model (right click on model - Simplify model), that worked fine for me.

lukasmatena, 2022-05-17:
@natehansen66 It works if I repair the model by NetFabb. The model is probably broken.

neophyl, 2022-05-17:
I've had issues with the paint tool not being able to pain faces when there were intersecting face errors on a model.  It couldn't select the ones 'behind' as such.  When I fixed the errors the paint tool worked afterwards.  Sounds like what is happening on this one.

natehansen66, 2022-05-17:
@lukasmatena this was a mesh created by Solidworks so not surprised it's broken. I forget about NetFabb that does indeed do the trick. Thanks!

natehansen66, 2022-05-17:
Should I close this?

rtyr, 2022-05-17:
I believe we can close this.

closed 2022-05-17 by rtyr