prusa3d/PrusaSlicer #8332: Associate .3mf files with PrusaSlicer in Windows

ygoe, 2022-05-14:
### Description of the bug

.3mf files are not opened with any application in SpeedCommander. They also have no icon. When double-clicking, I get the error message that there is no application associated with this file type. When double-clicking in Windows Explorer, it opens in Paint 3D. I guess that's an error.

I already checked the program settings and enabled the corresponding option to associate PrusaSlicer with these formats. The setting is also saved for the next program start. But it has zero effect.

### Project file & How to reproduce

All .3mf files

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2+win64

### Operating system

Windows 10 21H2

### Printer model

Mini+ (irrelevant)

stepikovo, 2022-05-15:
Have you checked the settings in Windows? https://support.microsoft.com/en-us/windows/change-default-programs-in-windows-e5d82cad-17d1-c53b-3505-f10a32e1894d

It should look like this then and works fine
[image]

bubnikv, 2022-05-16:
One may certainly register PrusaSlicer manually. We implemented automatic registration into PrusaSlicer Windows build, however it may not work correctly as it manipulates registry the old way.

We should do https://techcommunity.microsoft.com/t5/ask-the-performance-team/how-to-configure-file-associations-for-it-pros/ba-p/1313151

@enricoturri1966 is responsible for that.

ygoe, 2022-05-21:
In the Windows settings, both file extensions had no default app. I could select PruseSlicer from the list and now they open as expected. That's how I do it manually.

And what does the option in the PrusaSlicer settings do?

bubnikv, 2022-05-23:
We investigated the issue further. Since Windows 10 applications are no more allowed to register file extensions. Thus we will have to remove the option from PrusaSlicer settings.

We are investigating futher whether the installer will be allowed to register the file extension.

ygoe, 2022-05-23:
This isn't true, lots of other applications can do it as well. Maybe it's the API that has changed. Note that file extensions can be registered per system (required admin rights) or per user (works anytime).

bubnikv, 2022-05-23:
> This isn't true, lots of other applications can do it as well.

Which applications?

Two of us PrusaSlicer developers believe that Microsoft made a lot of
effort to disable such functionality. We will be most happy if you point us
to the API we should use.

> Note that file extensions can be registered per system (required admin
rights) or per user (works anytime).

By the application or by the user?

AFAIK it is not possible by the application. Yes, you can register the
extension manually.

ygoe, 2022-05-23:
Which applications can register their file extensions? At least in a way that Windows asks next time and suggests the new app in a list of alternatives? Umm, let me go through my installed applications...

* Firefox (.html, .url) (same for Brave and Chrome if you wanted)
* Visual Studio (.sln, .csproj, .cs and loads more)
* DBeaver (.sql)
* Inno Setup / Inno Script Studio (.iss)
* OpenSCAD (.scad) (I think FreeCAD also has one, I haven't really used it yet)
* Microsoft Office (.docx, .xlsx and loads more) (LibreOffice should also have a few)
* Acrobat Reader (.pdf with some strange instructions)
* Paint.NET (.png, .jpg ...)
* Affinity Photo/Designer (.afphoto, .afdesign)
* VLC (.mkv, .mp4 ...)
* Notepad++ (.txt and anything you want, with manual setup within the app settings, not Windows)

I usually open files through a file commander (like Explorer) and not through the app itself, so I know when it doesn't work.

That's the most common apps. I don't know how they do it. Most of them are open source. Maybe you can investigate how Inno Setup does it, IIRC it has a command to register file extensions for the installed app. Also, Stack Overflow is a good resource to ask for advice. It's been a while since I last registered a file extension for one of my apps, but it did work in Windows 10 as before without problems.

After all, there's a registry branch that contains all the data. You can always write it yourself: HKEY_CURRENT_USER\SOFTWARE\Classes

This should work:

```reg
Windows Registry Editor Version 5.00

[HKEY_CURRENT_USER\SOFTWARE\Classes\.3mf]
@="Prusa.Slicer.1"

[HKEY_CURRENT_USER\SOFTWARE\Classes\Prusa.Slicer.1\Shell\Open\Command]
@="\"C:\\Program Files\\Prusa3D\\PrusaSlicer\\prusa-slicer.exe\" \"%1\""
```

And here's the Inno Setup script to register *.fl with my app. Seems to use the system-wide settings.

```innosetup
[Registry]
; Register .fl file name extension
Root: HKCR; Subkey: ".fl"; ValueType: string; ValueName: ""; ValueData: "FieldLogFile"; Flags: uninsdeletevalue 
Root: HKCR; Subkey: "FieldLogFile"; ValueType: string; ValueName: ""; ValueData: "FieldLog file"; Flags: uninsdeletekey
Root: HKCR; Subkey: "FieldLogFile\DefaultIcon"; ValueType: string; ValueName: ""; ValueData: "{app}\FieldLogViewer.exe,1"
Root: HKCR; Subkey: "FieldLogFile\shell\open\command"; ValueType: string; ValueName: ""; ValueData: """{app}\FieldLogViewer.exe"" ""%1"""
Root: HKCR; Subkey: "FieldLogFile\shell\opensingle"; ValueType: string; ValueName: ""; ValueData: "{cm:OpenSingleFileCommand}"
Root: HKCR; Subkey: "FieldLogFile\shell\opensingle\command"; ValueType: string; ValueName: ""; ValueData: """{app}\FieldLogViewer.exe"" /s ""%1"""

; Add to .fl "Open with" menu
Root: HKCR; Subkey: ".fl\OpenWithList\FieldLogViewer.exe"; ValueType: string; ValueName: ""; ValueData: ""; Flags: uninsdeletekey
Root: HKCR; Subkey: "Applications\FieldLogViewer.exe"; ValueType: string; ValueName: "FriendlyAppName"; ValueData: "FieldLogViewer"; Flags: uninsdeletekey
Root: HKCR; Subkey: "Applications\FieldLogViewer.exe\shell\open"; ValueType: string; ValueName: "FriendlyAppName"; ValueData: "FieldLogViewer"
Root: HKCR; Subkey: "Applications\FieldLogViewer.exe\shell\open\command"; ValueType: string; ValueName: ""; ValueData: """{app}\FieldLogViewer.exe"" ""%1"""
```

bubnikv, 2022-05-23:
It may be that it was not the application, but the application installer
that did the registration. We are looking into it.
Applications are not allowed to change registration anymore. Installers
likely are. PrusaSlicer team is only writing the application while another
team is doing the installer for us. We are working on that.

bubnikv, 2022-05-23:
> After all, there's a registry branch that contains all the data. You can
always write it yourself: HKEY_CURRENT_USER\SOFTWARE\Classes

Since Windows 10, this section needs to be signed and the code to sign the
section is not public. The manual registration does not work if it is not
signed properly.

ygoe, 2022-05-23:
Hm, I thought that installers were just applications, too. (I'm talking about Inno Setup, Nullsoft Installer or other custom installers, not that MSI crap.) The registry API is available to all processes, within their user account privilege limits. And the current_user hive is certainly writeable by the user, that's what it's there for. It means that format associations are only set for the current user, not all users on the system, but that's totally fine actually. It doesn't matter if it's the installer or the app itself that does the association, it's just a matter of user experience. The installer provides a little more comfort and an easy start (and can register for all users if already installing for all users as admin); the app setting allows easy later changes. I'm fine with either one.

And tell me more about that "registry signing"? Never heard of that.

bubnikv, 2022-05-24:
I was as surprised as you are.

https://www.howtogeek.com/fyi/microsoft-broke-windows-10s-file-associations-with-a-botched-update/
https://kolbi.cz/blog/2017/10/25/setuserfta-userchoice-hash-defeated-set-file-type-associations-per-user/
http://woshub.com/managing-default-file-associations-in-windows-10/
https://devblogs.microsoft.com/oldnewthing/?p=96175
https://www.winhelponline.com/blog/windows-10-resetting-file-associations/

I have to digest it myself. Maybe they block just those extensions to be
modified which were already registered. For us it means we cannot change
registration of 3MFs, because they were associated with the microsoft 3D
Builder.

bubnikv, 2022-05-24:
https://docs.microsoft.com/en-us/archive/blogs/windowsinternals/windows-10-how-to-configure-file-associations-for-it-pros

bubnikv, 2022-05-24:
> Implementation of the hashing algorithm in some obscure basic-like language
> https://github.com/DanysysTeam/SFTA
>
>
> https://www.theverge.com/2021/9/13/22671182/mozilla-default-browser-windows-protections-firefox
> https://news.ycombinator.com/item?id=28510490&p=2
>
> You see, that's why we still don't have the FDM tree supports.

https://danysys.com/set-file-type-association-default-application-command-line-windows-10-userchoice-hash-internal-method/

Source code of the hash generator
https://pastebin.com/yVhWeQ3X

and how Mozilla did it
https://searchfox.org/mozilla-central/source/browser/components/shell/WindowsUserChoice.cpp#194
https://hg.mozilla.org/mozilla-central/rev/e928b3e95a6c3b7257d0ba475fc2303bfbad1874
https://hg.mozilla.org/releases/mozilla-release/diff/7e775ce432b599c6daf7ac379aa42f1e9b3b33ed/browser/components/shell/WindowsUserChoice.cpp

bubnikv, 2022-10-11:
Implementation of the hashing algorithm in some obscure basic-like language
https://github.com/DanysysTeam/SFTA

https://www.theverge.com/2021/9/13/22671182/mozilla-default-browser-windows-protections-firefox
https://news.ycombinator.com/item?id=28510490&p=2

You see, that's why we still don't have the FDM tree supports.

lukasmatena, 2023-02-01:
We tried to fix this in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Feedback is appreciated.

Ali229, 2023-02-02:
I'm not 100% if this is the same issue but I cannot open .stl files using prusa slicer 2.6.0-alpha2.
Reproduced by:
1. Right click stl file go to properties> change open with and set to prusa slicer 2.6.0-alpha2.
2. Open the stl file
3. Get prompt "Select an app to open this .stl file" (selecting prusa-slicer from the list shows this prompt again)

Not sure if this is caused by multiple installations (both alpha and stable), I tried removing others but issue persists.

closed 2026-09-29 by prusaslicer-bot[bot]