prusa3d/PrusaSlicer #8334: Two perimeters in one line

demonlibra, 2022-05-15:
### Description of the bug

With some perimeter width settings, two perimeters overlap into one line.

[image]
[image]
[image]

[image]

### Project file & How to reproduce

[Shape-Box.zip](https://github.com/prusa3d/PrusaSlicer/files/8694311/Shape-Box.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Windows 10, LinuxMint 20

### Printer model

UNI

demonlibra, 2022-05-15:
Te same problem with cilinder.

[image]
[image]

willkaxxx, 2022-05-21:
#8306 Copy

NGC869, 2022-06-21:
Hi, already mentioned here:
https://github.com/prusa3d/PrusaSlicer/issues/7973
https://github.com/prusa3d/PrusaSlicer/issues/7949

stepikovo, 2022-07-02:
We released [PrusaSlicer 2.5.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.5.0-alpha2) today with the Arachne perimeter generator, it should fix the issue, can you please confirm?

demonlibra, 2022-09-08:
Sorry for long reply. Dont get notify about your comment.
Checked in PrusaSlicer 2.5.0. Problen was resolved.
Thank you.

closed 2022-09-08 by demonlibra