prusa3d/PrusaSlicer #8338: Concentric Top Fill leaves gaps

cpotter485, 2022-05-17:
### Description of the bug

Making a gasket with NinjaFlex.  Used top layer concentric fill to improve the surface fill for a round container.  The slicer left a gap. in the top layer.  This is visible in the attached 3mf file. 
[Canister Gasket.zip](https://github.com/prusa3d/PrusaSlicer/files/8710258/Canister.Gasket.zip)
 

### Project file & How to reproduce

Designed simple ring in FreeCAD.  Saved and exported .stl file.
Loaded .stl into prusaslicer
Used custom print and filament settings I created for NinjaFlex.  (15mm/sec print speed and 230 C
set top fill to concentric and sliced.  

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Windows 10

### Printer model

Prusa MK3+

rtyr, 2022-05-17:
Same as https://github.com/prusa3d/PrusaSlicer/issues/1461. I am closing this as a duplicate.

closed 2022-05-17 by rtyr

cpotter485, 2022-05-17:
Thanks for the update.  My apologies.  I did a search for a duplicate before I submitted it.  But I guess I didn’t use the right key words.
 
And thank you for addressing the issue.  I complement you folks.  Prusa’s slicer is much better quality than others I have used.  
 
cp