prusa3d/PrusaSlicer #8340: Improve slicing resolution: The Benchy numberplate

plampix, 2022-05-19:
When I slice a Benchy using a .2 profile on PrusaSlicer, and compare the sliced preview to one I sliced in Cura, the PrusaSlicer gcode shows way less detail in the "#3DBenchy" numberplate. I have to reduce the outer perimeter width to .24 to get the preview to look as detailed as it does in Cura. Changing gcode or slicer resolution doesn't have an effect on this particular detail.

[Benchy_Sliced_by_Cura.gcode.gz](https://github.com/prusa3d/PrusaSlicer/files/8730561/Benchy_Sliced_by_Cura.gcode.gz)

I noticed this when I looked at a Benchy printed by someone else at .2 layer height, sliced in Cura, and I had never seen such a nice, legible text. I knew it wasn't because of the print quality, so some further investigation led me to Slicer differences
 
[image]

dartrax, 2022-05-25:
Did you try thin walls setting? May be related to #7530

plampix, 2022-05-25:
> Did you try thin walls setting? May be related to #7530

I did, it doesn't appear to be related.