prusa3d/PrusaSlicer #8342: Height Range Modifier Settings Closing Automatically

d3molite, 2022-05-20:
### Description of the bug

Every time a setting on a height range modifier is changed, the interface selection reverts back to the model and the settings are no longer shown requiring a click on the range again. This is frustrating when working with many range modifiers.

(Just before confirming the input)
[image]

(directly after confirming the input)
[image]

Expected behavior: 
Settings to not close until the user navigates somewhere else on his own.

### Project file & How to reproduce

Create a project and load a model with some instances. 

Apply a height range modifier to the object and add some settings that have more than one input box. 
[image]

Enter a setting. The interface will revert back to the default model selection.

[bug.zip](https://github.com/prusa3d/PrusaSlicer/files/8740367/bug.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2

### Operating system

Windows 11

### Printer model

Prusa i3 MK3S+ with MMU 2S

zbuilder801, 2022-06-02:
Not sure if this will help but... make sure the mouse cursor does not enter the Platter window; otherwise, when pressing the number keys other things may happen.

FidelCapo, 2022-06-17:
Thank you for report. This issue should be fixed in next release.
Closing.