prusa3d/PrusaSlicer #8343: between_object_gcode should be emitted in concurrent multipart prints

dotorg, 2022-05-20:
**Is your feature request related to a problem? Please describe.**
The current implementation of the code that handles "between_object_gcode" insertion only runs when using sequential printing. This severely limits the benefits of part tagging, as you can't using tooling or firmware support for canceling an object when you're printing concurrently. And, in fact, since a failed part in a concurrent print requires scrapping the entire print run, the limitation is far more severe, as opposed to sequential printing which doesn't waste the rest of the print if you were to stop it entirely. 

**Describe the solution you'd like**
The after_layer_change GCode should be inserted into the output between parts on every layer in non-sequential mode, or between parts in sequential mode. Then firmware/tools can cancel a specific part even in a concurrent multipart print. 

**Describe alternatives you've considered**
There really aren't any alternatives. Even a hacky post-process script to do it via the object label support won't work because *those* are also not output when printing multiple object concurrently.

(Which probably should also be changed along with this!)

neophyl, 2022-05-20:
Octoprint based prints can already cancel objects if the gcode has the verbose option enabled. That adds an object number to each line of gcode iirc. As octoprint spools out the lines to the printer bit by it can choose which ones not to send if you cancel an individual object. 

So the output from the slicer is already capable of supporting such functionality. 
Printer firmware would be the thing needing updates if it was to support it natively by looking for the tag on each line. 

For the Prusa line of printers I believe that feature has already been requested in the respective GitHub project. I doubt the 8bit boards have room left to implement it though. I could see the 32 bit ones like the mini being able to do it eventually if it’s something Prusa wants to expend resources on.

dotorg, 2022-05-20:
The slicer, again, only does that if you have it printing parts sequentially.  That's the issue -- the slicer does not actually insert the information into the gcode in normal print mode.

neophyl, 2022-05-20:
I just checked the latest 2.4.2 and it doesnt seem to label each line as before, I think I was misremembering the verbose option as its been ages since I used it, however it does similar if you have Print Settings>Output Options>Output File>Label Objects enabled.

For example with normal  non sequential printing (option=Complete Individual Objects = Off)  you get something like the following
G1 X157.831 Y92.056 E.03797 ; infill
; stop printing object Shape-Sphere id:1 copy 0
; printing object Shape-Box id:0 copy 0
G1 E-3.5 F3600 ; retract

Every time it swaps from one object to another on a layer you get similar messages between the blocks.  

I also just checked with the option for Complete Individual Objects=True and it still inserts object labels on each layer.  SO it looks like the option you want is called Label Objects

dotorg, 2022-05-20:
That is not the functionality I'm talking about. 

Its right in the subject -- "between_object_gcode". 

Labels are, at best, a poor hacky work-around. Comments are visual, and not meant to be parsed (as you noticed, how they are formatted has changed). That's why the slicer has actual built-in functionality (again, "between_object_gcode") to do it properly.

neophyl, 2022-05-20:
Ok.  Just pointing out that Octoprint manages to handle this functionality and has done for awhile with the current implementation.  IIRC its all handled by Octoprint and not post processed by the slicer.  Seems for it the labels are enough.

I'm sure the prusa devs will get around to making changes at some point.  As you say, nothing stays the same.  Good Luck.

closed 2026-09-29 by prusaslicer-bot[bot]